usnistgov/ARIAC #83: frame name protocol

dan9thsense, 2021-03-23:
Parts in logical cameras have frame names like:
`logical_camera_0_assembly_battery_green_1_frame`

and

`logical_camera_0_assembly_battery_green_3_frame`

Can you explain how the last digit increments?  It is not an integer corresponding to multiple parts in a camera, since something like

`logical_camera_0_assembly_battery_green_3_frame`

can show up even if there is only one part in the camera view.  Is it the number of times that the camera has seen a part type?

zeidk, 2021-03-23:
Hi Dan, I need to take a look at the config file you are using. Can you send it to me? There may be already two green batteries in the environment which are not in the camera frame.

dan9thsense, 2021-03-24:
Ah, so the second number may iterate when a part of the same type is seen in a different camera?  OK, I will send the config file.

denbar-robotics, 2021-03-30:
Thanks @stevegray954   Does that part name persist as the part is moved around?  Does that mean putting a part from the belt and a part from a bin onto the same agv could result in two parts with the same frame name?  That can't be good.

stevegray954, 2021-03-30:
Yes, that part name persists.

zeidk, 2021-04-20:
I have taken care of not having parts with the same ID when the belt is used. If you have 4 blue batteries in a bin (battery_blue_1,..., battery_blue_4) then the first blue battery on the belt will start with ID=5. In previous versions the first blue battery started with ID=0.

closed 2021-04-20 by zeidk

denbar-robotics, 2021-04-20:
If a part that was previously not under a camera is brought under a logical camera, does it get assigned the next unused ID?
And that would presumably cause a skip in the ID numbers on the belt, is that right?

stevegray954, 2021-04-20:
The names and ID numbers have nothing to do with the cameras themselves,
they are assigned when the object is created in Gazebo.

denbar-robotics, 2021-04-20:
We were getting duplicate numbers on the belt so I am wondering if that might happen when a new object is seen elsewhere.