usnistgov/ARIAC #84: Unable to submit shipment

GerardHarkemaAvans, 2021-03-25:
Hello,

when i commit a shipment with "rosservice call /ariac/agv2/submit_shipment "as1" "order_0_kitting_shipment_0"" the agv does not move to assembly station 1. Also there is no scoring message in the terminal of the sample_environment (i saw that in previous versions of Ariac).

Output of the 'command" terminal:
gerard@gerard-HP-ProBook-440-G6:~/ariac2021_ws$ rosservice call /ariac/agv2/submit_shipment "as1" "order_0_kitting_shipment_0"
success: True
message: ''
gerard@gerard-HP-ProBook-440-G6:~/ariac2021_ws$ clear

gerard@gerard-HP-ProBook-440-G6:~/ariac2021_ws$ rosservice call /ariac/start_competition
success: True
message: "competition started successfully! GOOD LUCK!"
gerard@gerard-HP-ProBook-440-G6:~/ariac2021_ws$ rosservice call /ariac/agv2/submit_shipment "as1" "order_0_kitting_shipment_0"
success: True
message: ''
gerard@gerard-HP-ProBook-440-G6:~/ariac2021_ws$ 

Output of the environment terminal:
[ WARN] [1616697009.832988247, 9.919000000]: Force enabling conveyor because power set by task manager.
[ INFO] [1616697009.844375039, 9.920000000]: Announcing order: order_0

[ WARN] [1616697015.431746618, 11.095000000]: [INFO] AGV 'agv2' tasked to go to AS1
[ INFO] [1616697015.431864959, 11.095000000]: [agv2-> as1] SUCCESS: AGV successfully triggered.

Kind regards Gerard

zeidk, 2021-03-26:
This should have been fixed in the latest update. I will take another look at what is happening.

GerardHarkemaAvans, 2021-03-26:
Sorry,

forget to build the workspace. It works now.

Thanks,

Gerard

closed 2021-03-26 by zeidk