usnistgov/ARIAC #86: [ERROR] : Action client not connected: when rgbd_camera is used

syncsyncsync, 2021-04-05:
When I try to use rgbd_camera  and start the simulation, I have an error as below.
I found this  error happens only when I use rgbd_camera and other sensors are available with no error. 
This may just depend on my test environment, but could you please investigate if it can be reproduced?

INFO] [1617606864.349645, 0.262000]: Calling service /gazebo/set_model_configuration
[ WARN] [1617606864.352361800, 0.263000000]: setModelConfiguration service called
[INFO] [1617606864.356225, 0.263000]: Set model configuration status: SetModelConfiguration: success
[ariac/kitting/kitting_spawner-13] process has finished cleanly
[ INFO] [1617606865.663275600, 0.530000000]: Camera Plugin: Using the 'robotNamespace' param: '/ariac/gantry/'
[ INFO] [1617606865.664844100, 0.530000000]: Camera Plugin (ns = /ariac/gantry/)  <tf_prefix_>, set to "/ariac/gantry"
[ WARN] [1617606872.671873700, 5.014000000]: Waiting for /ariac/gantry/gantry_arm_controller/follow_joint_trajectory to come up
[ WARN] [1617606878.917111500, 11.014000000]: Waiting for /ariac/gantry/gantry_arm_controller/follow_joint_trajectory to come up
[WARN] [1617606884.823010, 16.543000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1617606884.881239, 16.599000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[ariac/gantry/joint_state_controller_spawner-6] process has finished cleanly
[ariac/gantry/gantry_controller_spawner-9] process has finished cleanly
[ERROR] [1617606885.315831100, 17.014000000]: Action client not connected: /ariac/gantry/gantry_arm_controller/follow_joint_trajectory
[ WARN] [1617606890.635545200, 22.110000000]: Waiting for /ariac/gantry/gantry_controller/follow_joint_trajectory to come up
[ WARN] [1617606896.892778800, 28.111000000]: Waiting for /ariac/gantry/gantry_controller/follow_joint_trajectory to come up
[ERROR] [1617606903.190334600, 34.111000000]: Action client not connected: /ariac/gantry/gantry_controller/follow_joint_trajectory
[ INFO] [1617606903.228066700, 34.147000000]: Returned 0 controllers in list
[ INFO] [1617606903.236632900, 34.154000000]: Trajectory execution is managing controllers

I'm using fixed_sample_environment.launch with default sample_kitting.yaml.
My user_config (yaml) that is below.

sensors:
  rgbd_camera0:
    type: rgbd_camera
    pose:
      xyz: [-1.8, 0, 3.000004]
      rpy: [3.141593, 0.000000, 1.5707923]

Thank you in advance.

zeidk, 2021-05-04:
We have the following errors with sample_user_config_rgbd.yaml. The errors are different from the ones posted by @syncsyncsync though.

[INFO] [1620137678.452069, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1620137678.460432, 0.000000]: Calling service /gazebo/spawn_urdf_model
[ INFO] [1620137679.144966284]: Camera Plugin: The 'robotNamespace' param did not exit
[ INFO] [1620137679.146785369]: Camera Plugin: The 'robotNamespace' param did not exit
[ERROR] [1620137679.146892398]: Tried to advertise a service that is already advertised in this node [/set_camera_info]
[ INFO] [1620137679.147943189]: Camera Plugin: The 'robotNamespace' param did not exit
[ERROR] [1620137679.148022667]: Tried to advertise a service that is already advertised in this node [/set_camera_info]
[ INFO] [1620137679.148047459]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[ WARN] [1620137679.148072092]: dynamic reconfigure is not enabled for this image topic [ariac/rgbd_camera_bins0/ir/image_raw] becuase <cameraName> is not specified
[ INFO] [1620137679.148141533]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[ WARN] [1620137679.148179887]: dynamic reconfigure is not enabled for this image topic [ariac/rgbd_camera_bins1/ir/image_raw] becuase <cameraName> is not specified
[ INFO] [1620137679.149411273]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[ WARN] [1620137679.149427629]: dynamic reconfigure is not enabled for this image topic [ariac/rgbd_camera_station4/ir/image_raw] becuase <cameraName> is not specified
[ INFO] [1620137679.170147122]: Camera Plugin: The 'robotNamespace' param did not exit
[ERROR] [1620137679.170268942]: Tried to advertise a service that is already advertised in this node [/set_camera_info]
[ INFO] [1620137679.171840530]: Camera Plugin: The 'robotNamespace' param did not exit
[ INFO] [1620137679.172742154]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[ WARN] [1620137679.172764075]: dynamic reconfigure is not enabled for this image topic [ariac/rgbd_camera_station2/ir/image_raw] becuase <cameraName> is not specified
[ INFO] [1620137679.173175111]: Camera Plugin: The 'robotNamespace' param did not exit
[ERROR] [1620137679.173383403]: Tried to advertise a service that is already advertised in this node [/set_camera_info]
[ERROR] [1620137679.175670888]: Tried to advertise a service that is already advertised in this node [/set_camera_info]
[ INFO] [1620137679.175916760]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[ WARN] [1620137679.175940340]: dynamic reconfigure is not enabled for this image topic [ariac/rgbd_camera_station3/ir/image_raw] becuase <cameraName> is not specified
[ INFO] [1620137679.177646729]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[ WARN] [1620137679.177663063]: dynamic reconfigure is not enabled for this image topic [ariac/rgbd_camera_station1/ir/image_raw] becuase <cameraName> is not specified
[ WARN] [1620137679.403392847]: spawnURDFModel service called
[ WARN] [1620137679.403511163]: spawnSDFModel service called

zeidk, 2022-03-28:
Closing for now. Reopen if needed.

closed 2022-03-28 by zeidk