usnistgov/ARIAC #87: start tolerance

denbar-robotics, 2021-04-06:
If the arm gets deviated, for example, by pushing too hard onto a part, then the next move often fails with this message:

[ERROR] [1617674920.703818287, 50.648000000]: 
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'wrist_3_joint': expected: 0.0132668, current: 0.00210429

I tried setting the parameter in a launch file l(see line below) and checking `param get`, it says it was changed, but the error remains.
`<param name="/ariac/kitting/move_group/trajectory_execution/allowed_start_tolerance" value="1.0"/>`

Is there some way to increase this tolerance?  Or is this from a different parameter?

denbar-robotics, 2021-04-21:
This happens with moveit pose moves.  It occurs with the gantry arm a lot and the kitting arm occasionally.  It leads to unexpected aborts of the path plan.

Here is a recent example from the kitting arm:

[ERROR] [1618962169.632292069, 111.289000000]: 
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'wrist_3_joint': expected: 4.1384, current: 4.12603

johnmichaloski, 2021-04-21:
I get this when the KDL kinematic solver changes the robot configuration such as an elbow flip (for other ROS applications). Really messes up the Cartesian pose trajectory planner. Isn't supposed to do it, as KDL is supposed to pick the nearest joint solution.

Pavel has suggest reducing the joint limits.

denbar-robotics, 2021-04-21:
Note that it occurs far from joint limits, so seem like the limit may not be the issue.  There is a parameter that seems like it should help, but setting it to 1.0 does not make any difference:
```
<param name="/ariac/kitting/move_group/trajectory_execution/allowed_start_tolerance" value="1.0"/> 

<param name="/ariac/gantry/move_group/trajectory_execution/allowed_start_tolerance" value="1.0"/>
```

johnmichaloski, 2021-04-21:
I meant if you doing a straight line cartesian motion, you might have a elbow flip (quite common) which really messes up moveit pose trajectory.  Plus you might not see it in a simulation since the flip can be one cycle (if you get away with it). computeCartesianPath not happy large jump_threshold.

           // jump_threshold No more than jump_threshold is allowed as change in
            // distance in the configuration space of the robot
            // (this is to prevent 'jumps' in IK solutions). 0.0 disables

denbar-robotics, 2021-04-21:
Oh, I see. interesting.  Thanks for the insight.

denbar-robotics, 2021-06-02:
This still happens a lot.......

mcboyd-bu, 2021-06-02:
Agreed. Experiencing the same issue. Thought it was just us. Already setting jump_threshold to 0.0. Not sure how else to address it.

denbar-robotics, 2021-06-02:
@mcboyd-bu that is a good idea-- does it really allow the trajectory to proceed?

This is a bad bug because it kills the trajectory unexpectedly and the aborts are problematic for the code.

mcboyd-bu, 2021-06-02:
Everything proceeds normally...until the times when the trajectory is
aborted.

Using this value of 0.0 has no discernable impact on this specific bug, but
wanted to make sure that fact was added to the dialog here.

glpuga, 2021-06-02:
I had some issues like this when transitioning from one trajectory plan to another, especially with the gantry robot, because the robot would oscillate slightly when stopping suddenly, and the start pose sampled when starting planning would then differ a bit from the actual pose when actual execution started.

In my case it improved a bit by limiting the maximum speed and acceleration of the gantry joints. Some arm poses also worked better than others when carrying parts.

denbar-robotics, 2021-06-02:
I originally thought that the residual movement were causing it too.  However, it seems to occur in the wrist joints which suggests it is not an overall pose issue.  For a while there were bugs in the gantry robot that caused a lot of collisions, especially when carrying parts, but I think those have been resolved.

AravindaDP, 2022-05-07:
This issue occurs time to time with `kitting_commander_node.py` (Based on which I implemented my solution) (Most probable cause is if kitting arm slacked and hit the conveyor before competitor solution started.)

Will it make sense if it can be ensured from competition server side to start the simulation so that kitting arm is not in collision with conveyor? Or is there a way to resolve this from competitor solution side? (May be increase threshold? Is it in a rosparam? or different API than using MoveGroupCommander go API call used)

denbar-robotics, 2022-10-11:
It still happens, but not nearly as often.