usnistgov/ARIAC #88: do the agvs move to the assembly stations?

denbar-robotics, 2021-04-07:
Testing the config file `sample_multiple_non_high_priority_orders.yaml` the AGV doesn't move when a shipment is submitted and also the assembly shipment is not triggered.  The shipment submitted response is `true` so it seems the submit service worked ok.

zeidk, 2021-04-08:
Hello, can you tell me the different service calls you are using and in which order you are calling these services?

denbar-robotics, 2021-04-08:
The transport service is `ariac/agv1/to_as1`
Just sending a trigger via `ServiceProxy` and getting the result True

denbar-robotics, 2021-04-08:
Also, it scores OK with this submission if there are no additional orders.

denbar-robotics, 2021-04-10:
Figured it out.

denbar-robotics, 2021-04-10:
Well, not completely.  I can get the agv to move now but I am unclear on when to do that.  It will only move if the kitting shipment is not submitted.  

I have an order that does kitting on agv2 with as1 as the assembly location, then does assembly from as1.  After completing the kitting, I submitted the shipment with:
rosservice call /ariac/agv2/submit_shipment "as1" "order_0_kitting_shipment_0"
It returns:
success: True
message: ''

I expected agv2 to move to as1, but it does not.  It just sits at ks2 and the parts lift off the tray and vibrate.  The sim console does not acknowledge the submission.  This looks just like issue #84 but I did rebuild the workspace and still have this problem.

If I try to move it or resubmit it, I get this error:

[ WARN] [1618024265.581718224, 43.157000000]: [INFO] AGV 'agv2' tasked to go to AS1
[ERROR] [1618024265.581774990, 43.157000000]: [agv2 already at as1] FAILURE: AGV not successfully triggered.

If the order just has a kitting task, then the submission works and scores correctly, although the agv still does not move.

zeidk, 2021-04-10:
I am trying to reproduce it on my side but it's working. I will grab the github version and try with this one. Yes the AGV should move to as1. You are doing it right. The failure is due to the topic having as1 being the current location for agv2.

denbar-robotics, 2021-04-10:
If submitting a kitting shipment automatically makes the agv move, then when would one need the service that makes it move?  Also, I noticed that the topic `/ariac/agv2/station` is not being published, so how can we ensure in our code that the agv station has moved to where it is ready for an assembly task?

denbar-robotics, 2021-04-11:
Another problem encountered.  After submitting a kitting shipment (on agv3 at ks3) the whole sim becomes unresponsive.  The assembly gantry and arm stop moving.  After a while, this message shows up in the console that is working on the gantry and gantry arm:

[ INFO] [1618114243.962362716, 70.532000000]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1618114243.962412310, 70.532000000]: Failed to fetch current robot state

The sim console is just frozen without any indication of a new error.

denbar-robotics, 2021-04-11:
Update-- running over and over, looks like the sim gets way overloaded after a kitting submission. The realtime factor drops from around 0.8 to 0.02.  This does not occur after an assembly submit.

denbar-robotics, 2021-04-11:
The log file has thousands and thousands of the same message:

assembly_regulator_red_1__joint__
(1618115691 129705776) [Dbg] [ROSAriacKitTrayPlugin.cc:320] Creating fixed joint: kit_tray_3_bin5|assembly_sensor_blue_1__joint__
(1618115691 130275214) [Dbg] [ROSAriacKitTrayPlugin.cc:320] Creating fixed joint: kit_tray_3_bin2|assembly_battery_blue_2__joint__
(1618115691 224525754) [Dbg] [ROSAriacKitTrayPlugin.cc:355] agv3::kit_tray_3::kit_tray_3::tray: Handle clear tray service called.
(1618115691 224548679) [Dbg] [ROSAriacKitTrayPlugin.cc:313] Number of models in contact with the tray: 3
(1618115691 224584776) [Dbg] [ROSAriacKitTrayPlugin.cc:320] Creating fixed joint: kit_tray_3_bin1|
assembly_regulator_red_1__joint__
(1618115691 225320716) [Dbg] [ROSAriacKitTrayPlugin.cc:320] Creating fixed joint: kit_tray_3_bin5|assembly_sensor_blue_1__joint__
(1618115691 225945920) [Dbg] [ROSAriacKitTrayPlugin.cc:320] Creating fixed joint: kit_tray_3_bin2|assembly_battery_blue_2__joint__
(1618115691 325452931) [Dbg] [ROSAriacKitTrayPlugin.cc:355] agv3::kit_tray_3::kit_tray_3::tray: Handle clear tray service called.
(1618115691 325476820) [Dbg] [ROSAriacKitTrayPlugin.cc:313] Number of models in contact with the tray: 3
(1618115691 325514702) [Dbg] [ROSAriacKitTrayPlugin.cc:320] Creating fixed joint: kit_tray_3_bin1

zeidk, 2021-04-13:
Ok, I fixed this. I will push the new update once I'm done with making sure all the issues in the other tickets have also been fixed. Latest time frame for the new push is Thursday of this week.

barrygxwan, 2021-04-19:
we also meet this problem, when finished the order_0, and call service /ariac/agv2/submit_shipment "as1" "order_0_kitting_shipment_0" .  agv2 not move and become unstable in gazebo.

zeidk, 2021-04-26:
This has been fixed in the merge_test_april branch. Can you give it a try? I am planning to make the agvs move slower when traveling to stations as2 and as4. Will this affect any of your current implementation?

barrygxwan, 2021-04-27:
> This has been fixed in the merge_test_april branch. Can you give it a try? I am planning to make the agvs move slower when traveling to stations as2 and as4. Will this affect any of your current implementation?

we have tried the merge_test_april branch, it works. by the way, how to enter the qualifier competition?  today is 26, april

zeidk, 2021-05-02:
Details about the qualifiers will be pushed very soon (probably today).

closed 2021-05-02 by zeidk