usnistgov/ARIAC #89: How to enable robot_camera?

Rohithv-r, 2021-04-07:
After the new updates, the option to enable the gantry robot_camera in the user configuration file by adding :
`robot_camera:
  enable: true`
doesn't seem to work. It gives the error 
**Error: unknown top level entry 'robot_camera'**

However, I can see that an option ` enable_robot_camera: true` has been added in the trial configuration files. This seems to drastically slow down the simulation startup (Takes several minutes to start, and even after gear.py is killed, the system becomes a bit slower for some time). This doesn't happen when _enable_robot_camera_ is set to _false_ . GEAR used to be much faster with the gantry camera enabled even in the March 13th update. 

Also how do competitors add / enable the gantry robot camera through the user configuration files? 

Thanks in advance!

zeidk, 2021-04-08:
Hello, the camera should be activated in the user configuration file. I will take a look and see if anything broke this feature.  enable_robot_camera should not be used in the trial configuration files anymore. This should be a quick fix.

zeidk, 2021-04-20:
Hello, sorry for the late reply. Can you switch to the merge_test_april branch and look here https://github.com/usnistgov/ARIAC/blob/2c56b117a542d9ee579ce6b1d3bec9fb0874517a/nist_gear/config/user_config/sample_user_config_robot_camera.yaml#L15

zeidk, 2021-04-20:
I'm closing this ticket for now. Feel free to ping me if you still can't get it to work.

closed 2021-04-20 by zeidk