usnistgov/ARIAC #90: kitting arm initial joints

denbar-robotics, 2021-04-07:
The kitting arm initial joints change from one run to the next.  Some of them are clearly wrong, as when it is bent over the rail.
The joint wrist1 joint seems to be off most frequently, varying from -2.7 to + 3.6.
Are there defined joints for the initial position of the arm?  If so, will contest runs that have it starting with incorrect joints be rerun?

zeidk, 2021-04-08:
Do all the controllers start when you start the simulation? I have not noticed this at all and I'm curious as to how you have this behavior. Do you mind uploading a log file?

denbar-robotics, 2021-04-08:
Sure, which log file do you want?  All the controllers load OK.  There are no errors in performance.log  This happens about 20% of the time.  The arm works OK, it would just be nice to know where it is going to start.

I'm attaching an image as an example.  Here are the joint states:

```
header: 
  seq: 4995
  stamp: 
    secs: 99
    nsecs: 922000000
  frame_id: ''
name: [elbow_joint, linear_arm_actuator_joint, shoulder_lift_joint, shoulder_pan_joint,
  vacuum_gripper_joint, wrist_1_joint, wrist_2_joint, wrist_3_joint]
position: [-0.6114049668683714, 0.014174808482274743, 0.3009926442078008, -0.043833719881464894, -7.716944956825955e-06, -0.18999951386964886, 0.5113663472643486, 1.0543961338560592]
velocity: [-0.3611036343328132, -0.00275868007088335, 0.11553725374175966, 0.00322258210696799, -0.001824431520466836, 0.10673331343770576, -0.0038725146651870015, 0.05741614411719662]
effort: [-76.63983908208905, -37.06571039422064, 330.0, 38.91498422065733, 0.0, 0.29486501465384285, -0.00857803806322281, 0.04827968001563422]
---
```
[image]

zeidk, 2021-04-10:
Ok I will look into it. The next push should clear all the tickets we have on Github. Does this happen very often? Once in a while? performance.log in your ~/.ariac/log folder should be enough.

zeidk, 2021-04-20:
When you get a chance try the new branch and see if you still get this behavior. It looks like the controllers were not completely started before the robots were spawned.

zeidk, 2021-04-26:
Can I close this ticket if you have not experienced this behavior in the test branch?

denbar-robotics, 2021-04-27:
It still happens.

zeidk, 2021-06-01:
I modified the launch file to have a smaller joint position for wrist_1. This should fix this issue.

zeidk, 2021-06-02:
Let me know if this still happens, otherwise I will close it.

denbar-robotics, 2021-06-02:
Haven't seen it happen with the new update.  Let's close it for now and reopen if it shows up again.

closed 2021-06-02 by denbar-robotics