usnistgov/ARIAC #91: I have debugged the coordinate information of the regulator in the briefcase, and the regulator's falling position is accurate, but after falling to the assembly position, it will jitter and fall. Excuse me, what's the situation.

fengyongchaoshen, 2021-04-14:


zeidk, 2021-04-16:
@Pavel-P Can you take a look at this asap?

zeidk, 2021-04-26:
We are still working on this one. As such, the qualifiers will not ask you to do assembly with parts that are placed on the sides of the ventilator. Only parts placed on the flat surface of the ventilator will be used (battery and pump).

fengyongchaoshen, 2021-04-27:
When will the preliminaries be held and have they been postponed?

zeidk, 2021-05-03:
Just to let you know that assembly used during the qualifiers will only use the pump and the battery and not the 2 other parts.

mqt0029, 2021-05-09:
Does this extends to kitting as well, or kitting still use all parts, just assembly will be using the pump and battery exclusively?

barrygxwan, 2021-05-23:
> Just to let you know that assembly used during the qualifiers will only use the pump and the battery and not the 2 other parts.

hi，zeidk.
       is this issue is fixed?

zeidk, 2021-05-24:
@Pavel-P Has the assembly plugin been fixed? If so, we will include the regulator and the sensor as part of the assembly.

closed 2022-03-27 by zeidk