usnistgov/ARIAC #93: joint starting tolerances

denbar-robotics, 2021-04-17:
Planned arm movements with the gantry arm often fail due to starting constraints, where one joint will be a little more than .01 out of position.  Is there some way to increase that tolerance?  It causes the entire path plan to abort unexpectedly.

denbar-robotics, 2021-04-17:
Looks like wrist3 is the worst offender.

zeidk, 2021-04-20:
Ok, I will look into this today. Are you using pose targets with moveit? I guess I can take a look at your code but not sure where to look. Also, using a different planner may give you better results but I doubt there will much improvement. I will look at the moveit package. Do you use the kitting robot? Have you experienced the same kind of issues?

denbar-robotics, 2021-04-20:
It almost never happens with the kitting robot arm unless it has previously collided with something.  Yes, pose targets with moveit.

denbar-robotics, 2021-04-20:
I just realized that I had previously posted this issue as #87 so I am closing this one.

closed 2021-04-20 by denbar-robotics