usnistgov/ARIAC #97: Technology post：The real time factor becomes very small and gazebo works slowly when running

barrygxwan, 2021-04-29:
I use gazebo 9.17.0, when I running the " merge_test_april" version,  gazebo works very slow.  Anyone meet this problem？

zeidk, 2021-04-29:
Is this true for all configuration files or for just one configuration file? What is your real time factor compared to your previous real time factor?

barrygxwan, 2021-04-30:
hello, zeidk,
 It's true for all configuration files. The previous real time factor is about 0.6, and now is about 0.3. My computer:
processor:Intel® Xeon(R) W-2245 CPU @ 3.90GHz × 16 
GPU: TITAN RTX/PCIe/SSE2
memory: 64G

zeidk, 2021-05-02:
Well, I can't think of any changes that could affect the real time factor. Do you do sensor processing with point clouds? Is your real time factor low from the beginning or when the robot starts moving?

barrygxwan, 2021-05-02:
We don't processing with point cloud,  when the robot starts moving, the real time factor become low

dan9thsense, 2021-05-02:
I just installed the new code posted today and my real time factor plummeted to between 0.02 and 0.06.
With the same trial config, it was running at about 0.35 on the branch code and around 0.5 on the original code.  Those were both fine, but the 0.06 factor is not usable.

zeidk, 2021-05-03:
Can you both check if gazebo_state_logging is set to true in the trial files. The trial files uploaded today are the same as before except for the option gazebo_state_logging. If it's present can you remove this line and try again? This option is important when running your code in docker but you can remove it during testing/debugging.

zeidk, 2021-05-04:
@barrygxwan  I need a little bit more information. Does the RTF drop when you issue the service /ariac/start_competition? Can you start the simulation and in another terminal start the competition through a ros service call? Let me know how much the RTF drops.

zeidk, 2021-05-04:
If you are just experiencing a low RTF in general, removing unnecessary models from the environment during development will help. I added a section [here](https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/competition_specifications.md#remove-models-from-worldsariacworldtemplate).

denbar-robotics, 2021-05-05:
The RTF when the sim loads varies from 0.9 to 0.3, depending on the number of objects instantiated.  Starting the competition with
`rosservice call /ariac/start_competition`
causes the RTF to immediately drop to 0.02 - 0.06. 

 `gazebo_state_logging` does not seem to matter.

barrygxwan, 2021-05-05:
> 
> 
> @barrygxwan I need a little bit more information. Does the RTF drop when you issue the service /ariac/start_competition? Can you start the simulation and in another terminal start the competition through a ros service call? Let me know how much the RTF drops.

The RTF droped from about 0.8 to 0.25 in my computer

zeidk, 2021-05-06:
I created a branch called qualifiers_testing in which I simplified the collisions of multiple parts. Can you try it and see if the RTF drop is less consequent?

db7296, 2021-05-07:
Tried the test branch just now.  It only helps a tiny bit.   RTF still drops to 0.07 from 0.38

dan9thsense, 2021-05-07:
The test branch only helps a tiny bit.  The RTF dropped to 0.07 from 0.38

zeidk, 2021-05-08:
Did you try to remove other aesthetics components? I can create a minimal world so you can try it.

mqt0029, 2021-05-09:
Oddly enough it happened to me over the previous commits.

When I reverted to 931d8743d2e93d5e349ac1493ff4ada12cdf484d (`git reset --hard <id>`) starting RTF is 0.99 and sometimes dips to 0.8

When I reverted to origin/master (`git reset --hard origin/master`) starting RTF is 0.6 and drops to 0.4 when competition starts.

zeidk, 2021-05-10:
Yes, disabling gazebo_state_logging should help but this makes little difference for some competitors. I've been profiling the code but nothing stands out. I will take a look at some plugin updates happening in the background.

dan9thsense, 2021-05-11:
Tried with and without development mode-- no difference.

barrygxwan, 2021-05-23:
I found that gazebo 9.16 is more better than gazebo 9.17 or 9.18.

zeidk, 2021-05-30:
It seems we have a decent real time factor in the latest push on master branch. Closing this ticket for now.

closed 2021-05-30 by zeidk