usnistgov/ARIAC #100: conveyor belt is very slippery

denbar-robotics, 2021-05-01:
When trying to pickup parts from the conveyor belt, they frequently slip out from under the gripper.  Is it possible to increase the friction on the conveyor belt?

zeidk, 2021-05-01:
Hi Dan, I am about to release everything for the qualifiers. I am monitoring the posted requests and issues and I will try to address them in the current code, including this one.

denbar-robotics, 2021-05-04:
It's not that big a deal.  Also, it is probably a bad idea to change a parameter this late in the game, as it might break other competitors code. Consider it something that might be good for next year.  One thing that might help is publishing the gripper status more often, as then the parts might not have a chance to slip.

closed 2021-05-04 by denbar-robotics

mqt0029, 2021-05-07:
Agreed with @denbar-robotics, let's not change anything. Closed #104 as well.