usnistgov/ARIAC #99: arm health question

denbar-robotics, 2021-05-01:
Can an arm become disabled during a run or are they only disabled from the start?  Looking at the examples, it seems that an arm's health is determined at the start:

    kitting_robot_health: 1 
    assembly_robot_health: 1

zeidk, 2021-05-02:
The robots' status are determined for each order. This [example](https://github.com/usnistgov/ARIAC/blob/master/nist_gear/config/trial_config/sample_multiple_non_high_priority_orders.yaml) shows two orders and you can see that the health is defined for the order. 

You can have, for example:
 - `order_0`:
```
kitting_robot_health: 1
assembly_robot_health: 1
```
- `order_1`:
```
kitting_robot_health: 0
assembly_robot_health: 1
```

The controllers will still be working but you are not supposed to use the robot for which the health is 0. Robot's health status is published on `/ariac/orders`.

There are some situations which can disable the robot (see [here](https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/agility_challenges.md#new-task)). An example of trial file which describes this can be found [here](https://github.com/usnistgov/ARIAC/blob/master/nist_gear/config/trial_config/sample_task_changeover.yaml).

denbar-robotics, 2021-05-02:
Thanks for the fast and excellent answer.

closed 2021-05-02 by denbar-robotics