usnistgov/ARIAC #101: Question on Qualifiers Part A qual1.yaml

barrygxwan, 2021-05-02:
hello, is the config file qual1.yaml has problem?
**One order with 2 shipments. Both shipments need to be built on the same agv (agv3).**
    kitting:
      shipment_count: 2
      agvs: [agv3, agv3]
      destinations: [as3, as4]
      products:
        part_0:
          type: assembly_battery_blue
          pose:
            xyz: [0.1, 0.1, 0]
            rpy: [0, 0, 0]
        part_1:
          type: assembly_battery_green
          pose:
            xyz: [-0.15, -0.1, 0]
            rpy: [0, 0, 'pi/4']

this kitting task can not be accomplished, because After the first shipment agv3 _ as3 is submitted, we can not do kitting again on agv3, because agv3 is on as3, which violates the issue: https://github.com/usnistgov/ARIAC/issues/85

zeidk, 2021-05-02:
Yep, this is correct, thank you for pointing it out. The wrong file was submitted along with the wrong description. I will fix this right away.

barrygxwan, 2021-05-02:
by the way, we find the important dates are  changed,

    22-26 March 2021 - Smoke Test Week before Qualifier Round
    10-14 May 2021 - Qualifier Round
    17-21 May 2021 - Smoke Test before Final Round
    24-28 May 2021 - Final Round
So, the deadline for submit our files is 3 May 2021, or 14 May?

zeidk, 2021-05-02:
yes, this was explained in the updates in the wiki. Did you also receive an email? I guess it's a bit confusing but the deadline for submitting your code is May 10 (next Monday).

barrygxwan, 2021-05-02:
okay, we have reaceved发自我的荣耀手机-------- 原始邮件 --------发件人： Zeid Kootbally ***@***.***>日期： 2021年5月2日周日 半夜11:18收件人： usnistgov/ARIAC ***@***.***>抄送： barrygxwan ***@***.***>, Author ***@***.***>主    题： Re: [usnistgov/ARIAC] Question on Qualifiers Part A qual1.yaml (#101)
yes, this was explained in the updates in the wiki. Did you also receive an email? I guess it's a bit confusing but the deadline for submitting your code is May 10 (next Monday).

closed 2021-05-02 by zeidk