usnistgov/ARIAC #103: qual3 orders

dan9thsense, 2021-05-03:
The two orders for qual3 are sent simultaneously right at the beginning of the sim.  Even with setting a queue size = 3 for the order subscriber, often the first order is missed.  This is a known issue with ros, that the first messages in a topic can be missed if the subscriber loses the race to get subscribed in time and a second message is published right after the first.  Is it possible to delay one of the orders to give time for the subscriber to get established?

zeidk, 2021-05-03:
In `announcement_condition_value: 0.0`, 0.0 represents the simulation time after you start the competition (in this case the order is announced immediately). Can you set a different value and see if this helps? If it does then I think I will increase this time when multiple orders are required.

dan9thsense, 2021-05-03:
Changing the value to 3.0 fixes the issue.

zeidk, 2021-05-03:
Noted, I will proceed with this approach. Thanks.

closed 2021-05-03 by zeidk