usnistgov/ARIAC #104: gripper status lags

mqt0029, 2021-05-04:
when trying to pick up parts from the conveyor belt, often the part is vacuumed to the gripper, but the status is not updated until much later (GripperManager is_object_attached or directly subscribe to the topic), causing the belt to continue to push the part and force the arm to ABORT status due to the part dragging the arm out of alignment.

Is there any insight on this?

Note: this might be related to #100

zeidk, 2021-05-04:
Thank you for reporting this. Would you know the real time factor when this behavior was observed? I remember we had this issue in ARIAC 2020 with some competitors having a very low real time factor. I can probably increase the publishing rate and see if this helps.

mqt0029, 2021-05-04:
My real time factor is mostly stable at around 0.9, sometimes when the arm is moving it sometimes drops to around 0.8, but never below that. I don't think it can ever be flat at near 1 (e.g. 0.96-0.99) considering the amount of stuff present right? 

Also, are we allowed to adjust certain parameters such as publishing rate for certain publishers (i.e. gripper vacuum state) or joint tolerances?

zeidk, 2021-05-07:
You should do a pull request and I will see if it makes sense. Which publisher rates do you intend to adjust?

mqt0029, 2021-05-07:
I had assumed the topic `/ariac/kitting/arm/gripper/state` is responsible for telling if something is attached or not, which the rate is 1000 (reported by `rostopic hz`). To my understanding this is done via the Vacuum Gripper Plugin under `nist_gear/src/ROSVacuumGripperPlugin.cc`. 

At this point I am not sure if it's my implementation or something is off with the environment itself, because 1000 rate seems like it should have update the status as soon as it is attached. Then again, I am using Python, so it could just be the layering middleware that slows everything down.

closed 2021-05-07 by mqt0029

glpuga, 2021-05-19:
I'm looking into issues that seem related to this, and based on what I've been testing while the gripper state is published at a fast enough rate the plugin is only checking for contact at about 10 hz, which may be the main limitant here.

https://github.com/usnistgov/ARIAC/blob/master/nist_gear/src/VacuumGripperPlugin.cc#L561
https://github.com/usnistgov/ARIAC/blob/master/nist_gear/src/VacuumGripperPlugin.cc#L246

In my case, if the gripper is moving downwards when the contact is made, unless the gripper is moving really (really) slow, by the time the `attached` state is sampled by the plugin the gripper has already began forcing the wrist at an angle, and since the plugin will only declare contact if the normals are within 18 deg of each other, the `attached` flag never becomes becomes true and the movement eventually fails.