usnistgov/ARIAC #105: latest time for conveyor parts?

denbar-robotics, 2021-05-05:
If there is a part missing from the bins that is needed to fill an order, it may be possible to get it from the conveyor.  However, it is  not practical to wait forever for a part to show up on the conveyor.  At some point one has to ship.  Is there some amount of time by when conveyor parts will appear, say something like 30 seconds from the start?  That way, with incomplete orders, we do not have to wait around beyond that time, just in case a conveyor part might appear that could complete the order.

zeidk, 2021-05-05:
The time for the first part to show up on the belt is between 10 and 15 s. If after 15 s you don't see a part on the belt then the trial has no part on the belt. This information was never given to competitors in the past but I think knowing about this time value can help with the strategies. I will add this information to the wiki.

denbar-robotics, 2021-05-05:
That is very helpful, thanks.

I do have another question though.  In qual1, the first belt part is supposed to appear at 10 seconds, but it does not actually show up until 25 seconds in sim time (see output below).  Does the 15 second limit correspond to the value entered in the config file or the sim time when the part appears?  If it is the value in the config file, is there some corresponding sim time value to an config file entry of 15?

Here is the relevant section from the config file where it specifies the part appearing at 10 seconds.

belt_models:
  assembly_battery_green:
    10.0:
      pose:
        xyz: [0.0, 4.3, 0.92]
        rpy: [0, 0, 0]

And here is the relevant section from the performance log file, which is consistent with what happened during the run, where the part appeared at about the 25 second mark.

(1620189748 765466374) [Dbg] [ROSAriacTaskManagerPlugin.cc:1199] Sim time: 25
(1620189749 73740619) [Wrn] [Model.cc:703] Model::SetLinearAccel() is deprecated and has no effect. Use Link::SetForce() on the link directly instead. 
(1620189749 73767944) [Wrn] [Link.cc:733] Link::SetLinearAccel() is deprecated and has no effect. Use Link::SetForce() instead.
(1620189749 73787209) [Dbg] [PopulationPlugin.cc:406] Object [assembly_battery_green_1] on belt

zeidk, 2021-05-05:
Wow you are right, this number also varies, for you it was 25 but for me it was 35. I will try to track the issue. For now, you can try to replace 10 with a very low number like 2 or 3 and see if it shows up in the first 15 s after the competition starts.

denbar-robotics, 2021-05-05:
Setting the time to 2.0 causes the part to appear at about the 10 second time.  Setting it to 5.0 causes the part to appear at about 15 seconds.  One problem with the 2.0 value is that the subsequent parts come out every two seconds, so all the parts get used up quickly.  This is also an issue for the 5.0 time.   Increasing the number of parts fixes that, but also slows down the RTF.

zeidk, 2021-05-07:
My suggestion only works for the first part. I can spawn two types of part on the belt and I can make both types the same. Anyways, I think I tracked this delay and I should be able to fix this.

zeidk, 2021-05-10:
Fixed in the qualifiers branch.

closed 2021-05-10 by zeidk