usnistgov/ARIAC #106: qual1 question on conveyor parts

denbar-robotics, 2021-05-05:
In qual1, there is a blackout that occurs for most of the time that the conveyor parts are available.  By the time the blackout ends, there are no more parts being produced on the conveyor.  Is that intentional?  Along with having two of the first four conveyor parts be faulty, it leaves little time to gather them.

zeidk, 2021-05-05:
It was intentional but I think I will decrease the blackout time or not having blackout + parts on belt in the config files used in the qualifiers. I will reduce the length of the blackout for the current config file.

zeidk, 2021-05-05:
Also, if there is blackout + parts on belt I can increase the spawn time on the belt so that parts are still available on the belt when the blackout is over. Anyways, I will play with these values to make sure competitors can still access parts on the belt after a blackout.

closed 2021-05-30 by zeidk

denbar-robotics, 2021-05-30:
Is there now a maximum time to wait before knowing that a part will not appear on the belt?