usnistgov/ARIAC #107: gantry camera fails in docker

dan9thsense, 2021-05-07:
Trying to use the gantry camera in the docker file fails with this error:

[INFO] [1620359549.478755, 15.843000]: CvBridge Error: Unrecognized image encoding []
[INFO] [1620359549.479974, 15.844000]: failed to find an image from the gantry camera

When loading, docker has this error as well.  I tried installing opencv in my build bash file and it got rid of the error below but not the error above.

Warning [parser.cc:950] XML Element[plugin], child of element[camera] not defined in SDF. Ignoring[plugin]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
[Err] [DepthCameraSensor.cc:66] Unable to create DepthCameraSensor. Rendering is disabled.
[Err] [Sensor.cc:289] Sensor[gantry_tray_contact] is attempting to load a plugin, but detected an incorrect plugin type. Plugin filename[libROSAriacGantryTrayPlugin.so] name[gantry_tray_plugin]
[ WARN] [1620084076.157417735]: setModelConfiguration service called

zeidk, 2021-05-08:
Ok, I got it to work, there was some issues with the camera plugin not loaded in docker. I will push another version of the software as well as another image tomorrow.

dan9thsense, 2021-05-08:
Glad you got it to work!  I am concerned, however, that having it working tomorrow only leaves a single day for testing before the qualifiers deadline.

zeidk, 2021-05-30:
I am closing this now. "Unable to create DepthCameraSensor. Rendering is disabled." is expected since we are not running any x server in docker. Your issue is different and I will personally work with you once I release the new images.

closed 2021-05-30 by zeidk