usnistgov/ARIAC #108: Customizing moveit configuration files

glpuga, 2021-05-08:
Is it possible to use custom `gantry_moveit_config` and `kitting_moveit_config` configurations? If I'm following it correctly the evaluation system seems to load it's own set of configurations for this.

In particular, I'd like to be able to replace the kinematics solver for the `gantry_full` group, which is preconfigured for `cached_ik_kinematics_plugin/CachedSrvKinematicsPlugin` , which fails because it requires an external ik solver service.

Thanks!

zeidk, 2021-05-09:
Which kinematics solver are you planning to use? You can change Moveit parameters through your code but we are currently not allowing changes to the Moveit config files. At the same time we are not enforcing the use of Moveit and the packages provided were to help competitors who wanted to use Moveit. Could you do a pull request to see what changes you intend to do?

For the qualifiers everybody using Moveit will have to use the provided Moveit packages. If you are qualified and want to use a custom package for the finals I think we can modify the Docker scripts to pull a specific version of the Moveit packages.

glpuga, 2021-05-09:
[These are the changes](https://github.com/glpuga/ARIAC/commit/31935f3362e75bc4252ca41f25c780a42808c8e8) I'd like to be able to use (please see below that I don't need changes to the competition docker to use them). 

Mostly it's setting limiting the acceleration of the gantry and removing poses I did not use, but the one important change for me is [this one](https://github.com/glpuga/ARIAC/commit/31935f3362e75bc4252ca41f25c780a42808c8e8?branch=31935f3362e75bc4252ca41f25c780a42808c8e8&diff=unified#diff-2585714ab4863ec432d648b847bf3d5ac4dc0b10549e24ca86178801dc046239): 
```
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005
gantry_full:
- kinematics_solver: cached_ik_kinematics_plugin/CachedSrvKinematicsPlugin
+ kinematics_solver: kdl_kinematics_plugin/KDLKinematicsPlugin
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005 
```
That is because without that, trying to command the `gantry_full` with cartesian goal fails to plan with errors like the ones shown at the bottom of this post. The error seems to be because `CachedSrvKinematicsPlugin` depends on an external IK solver service not provided with the rest of the system. `gantry_arm` and `gantry_torso` use `kdl_kinematics_plugin/KDLKinematicsPlugin` instead, so I guess they don't fail.

> For the qualifiers everybody using Moveit will have to use the provided Moveit packages. If you are qualified and want to use a custom package for the finals I think we can modify the Docker scripts to pull a specific version of the Moveit packages.

I managed to get it to work with no changes to the `ariac-docker` by loading this configuration on a separate namespace and remapping the robot_description provided by the docker competition move_group instances. Here's the launch file that does it:

https://github.com/glpuga/ariac2021ws/blob/devel/project/the_italian_job/tijchallenger/launch/challenger.launch

Is this option valid for the qualifiers, given that it does not require any changes to the competition docker?

```
[ERROR] [1620600059.979836427, 81.343000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.980103514, 81.343000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
[ERROR] [1620600059.980117616, 81.343000000]: IK solver failed with error 99999
[ERROR] [1620600059.980135814, 81.343000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.980326891, 81.343000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
[ERROR] [1620600059.980342002, 81.343000000]: IK solver failed with error 99999
[ERROR] [1620600059.980357654, 81.343000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.980522592, 81.343000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
[ERROR] [1620600059.980535844, 81.343000000]: IK solver failed with error 99999
[ERROR] [1620600059.980552971, 81.343000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.980771815, 81.344000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
[ERROR] [1620600059.980786525, 81.344000000]: IK solver failed with error 99999
[ERROR] [1620600059.980797480, 81.344000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.980950984, 81.344000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
[ERROR] [1620600059.980961694, 81.344000000]: IK solver failed with error 99999
[ERROR] [1620600059.980972416, 81.344000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.981123487, 81.344000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
[ERROR] [1620600059.981134710, 81.344000000]: IK solver failed with error 99999
[ERROR] [1620600059.981149836, 81.344000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.981315266, 81.344000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
[ERROR] [1620600059.981326244, 81.344000000]: IK solver failed with error 99999
[ERROR] [1620600059.981340546, 81.344000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.981524037, 81.344000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
[ERROR] [1620600059.981536097, 81.344000000]: IK solver failed with error 99999
[ERROR] [1620600059.981549089, 81.344000000]: Seed state must have size 9 instead of size 13
[ERROR] [1620600059.981720374, 81.344000000]: Service call failed to connect to service: /ariac/custom/gantry/solve_ik
```

zeidk, 2021-05-10:
If the changes are made on your side and the current docker images stay intact then I don't see any problem here. Make sure you test everything in competition mode (remove --development-mode from sample_environment.launch) and see if the remapping performed in your launch file is not blocked in competition mode.

closed 2021-05-11 by glpuga