usnistgov/ARIAC #109: Unable to remove droped parts on qual1

glpuga, 2021-05-10:
On `qual1.yaml` I noticed that the robots were unable to remove the blue battery that is configured to be dropped on the tray [here](https://github.com/usnistgov/ARIAC/blob/qualifiers_testing/nist_gear/config/trial_config/qualifiers_practice/qual1.yaml#L99) after it had dropped on the tray. The robot would be unable to lift the part from the surface, despite repeated contact with the gripper on and multiple attempts.

After looking into I think the issue is [this line](https://github.com/usnistgov/ARIAC/blob/qualifiers_testing/nist_gear/config/trial_config/qualifiers_practice/qual1.yaml#L93) in `qual1.yaml`, which sets the min height to 0.0.

Changing that value for 0.1, the part still drops when above the tray, but now the robot has no problem removing it afterwards.

Is it possible that with the current configuration the part is being constantly being dropped from the gripper?

zeidk, 2021-05-10:
Does the part drop flat on the tray or at an angle? By changing the min height to 0.1 what difference do you see visually?

glpuga, 2021-05-10:
> Does the part drop flat on the tray or at an angle? 
Yes, the part falls flat. I tried making it fall on other places in the tray as well with the same result.

> By changing the min height to 0.1 what difference do you see visually?
I did not notice any difference visually after changing the value. Still the battery dropped flat, on the same spot, only that then the robot was able to pick it up afterwards.

glpuga, 2021-05-10:
I ran a few more experiments on this, and while I can still replicate not being able to lift the piece, I'm not so sure the drop configuration is a factor anymore. There is some random chance that grasping is successful, but I'm not sure what it correlates to.

Attached a video capture of two failed attempts to lift a piece, along with what the pose was after dropping. They are around 1:00 and 3:50.

https://user-images.githubusercontent.com/17802342/117654028-91c29e80-b16b-11eb-8803-b46bdb8edfbb.mp4

zeidk, 2021-05-10:
I know that sometimes if you push too hard on the part, the robot will not be able to pick up the part, especially for smaller parts like the battery. You can actually see around 1 min that one side of the part is lifted a little bit when you are picking up the part. Changing the value to 0.1 should not be the reason why you can/can't pick up the part because this is just used to define a region to trigger the drop. 

Do you always observe this behavior during a faulty gripper challenge? Could it be that your re-grasp strategy needs some tuning?
Which github branch are you using in the video?

glpuga, 2021-05-11:
>  I know that sometimes if you push too hard on the part, the robot will not be able to pick up the part, especially for smaller parts like the battery. You can actually see around 1 min that one side of the part is lifted a little bit when you are picking up the part. 

It's true. The behavior in the video is because the arm is moving downwards with a flat gripper at 0.25 cm steps, sensing the "attached" state of the gripper. Since the flag does not seem to become true, the arm continues lowering the gripper looking for contact.  The usual behavior is that the arm stops soon after coming in contact with the part.

> Do you always observe this behavior during a faulty gripper challenge? Could it be that your re-grasp strategy needs some tuning?

It's entirely possible that there's a problem with the strategy. The only thing that makes me think there's something different at play here is that this is the only scenario where I've seen this behavior. I've seen no significant issues grasping parts in other trial configurations (sample trial configs in the ARIAC repo + other two qualifier files) from bins or agvs. In this trial configuration, with that piece, it's pretty consistent.

I did not notice something like this when testing sample_faulty_gripper.yaml, but when I did the configuration was such that the parts would fall off the tray and no picking would be needed to remove them. I can try that and see what happens.

I tried changing the trial config to make the part drop at the center of the agv tray, to see if the fact that the part drops very close to the kitting rail caused gripper not to fall completely flat on the battery, but the result was the same.

> Which github branch are you using in the video?

I'm using the lastest master.

---

I brought this up in case someone else had noticed something similar. I'll continue looking into it. Thanks!

glpuga, 2021-05-14:
On a follow up to this, I set `verbose:=true` to see the debug out put of gazebo, and whenever the kitting robot fails to lift the part, the `attached` in the gripper state message never becomes true, and on the gazebo debug output, this begins to get printed repeatedly:

```
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin8|assembly_battery_blue_2::link::collision' and 'kitting::vacuum_gripper_link::vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin8|assembly_battery_blue_2
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin8|assembly_battery_blue_2::link::collision' and 'kitting::vacuum_gripper_link::vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin8|assembly_battery_blue_2
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin8|assembly_battery_blue_2::link::collision' and 'kitting::vacuum_gripper_link::vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin8|assembly_battery_blue_2
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin8|assembly_battery_blue_2::link::collision' and 'kitting::vacuum_gripper_link::vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin8|assembly_battery_blue_2
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
```

Nothing like this gets printed when lifting a non-dropped part.

Also, if the part dropped from the kitting robot, the kitting robot does not seem to have issue grasping the part and removing it. If the part gets dropped from the gantry robot, however, then the kitting robot is has the issue described above. This is weird, but pretty consistent.

Also, I've also seen this issue happen when running the configuration `sample_faulty_gripper.yaml`.

zeidk, 2021-05-17:
Awesome, thank you for tracking this down. I did made some modifications to the faulty gripper challenge and it may or may not have fixed this issue. I will check.

zeidk, 2021-05-21:
Thanks, the issue was related to an old ticket about different arms dropping the same part in the same tray, which was not the expected behavior. I will push a new release tomorrow (Friday) which fixes this issue.

zeidk, 2021-05-30:
Can you please re-test this with the most recent code on master branch?

glpuga, 2021-05-30:
I just checked the latest `master` with the same scenario (qual1.yaml). Unfortunately the issues seems to still be there:

- Gantry has no visible problem grasping a piece dropped from gantry.
- Kitting, however, is unable to grasp a piece dropped from gantry.

This is what a successful pick up looks like when seen on the gear logs with `verbose:=true`.

```
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 48
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 49
[Dbg] [ROSVacuumGripperPlugin.cc:113] Gripper control requested: Enable
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 50
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 51
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 52
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 53
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 54
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin3|assembly_battery_blue_7::link::collision' and 'gantry::gantry_arm_vacuum_gripper_link::gantry_arm_vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin3|assembly_battery_blue_7
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 55
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 56
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 57
```

This is what a failed pickup looks like when kitting tried to grasp a piece dropped by gantry:

```
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 150
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 151
[Dbg] [ROSVacuumGripperPlugin.cc:113] Gripper control requested: Enable
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 152
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 153
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin8|assembly_battery_blue_4::link::collision' and 'kitting::vacuum_gripper_link::vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin8|assembly_battery_blue_4
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 154
[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin8|assembly_battery_blue_4::link::collision' and 'kitting::vacuum_gripper_link::vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin8|assembly_battery_blue_4
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin8|assembly_battery_blue_4::link::collision' and 'kitting::vacuum_gripper_link::vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin8|assembly_battery_blue_4
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.
[Dbg] [VacuumGripperPlugin.cc:752] The gripper was in collision with its own model.
[Dbg] [VacuumGripperPlugin.cc:817] Number of collisions with gripper: 1
[Dbg] [VacuumGripperPlugin.cc:821] More collisions than the minContactCount: 1
[Dbg] [VacuumGripperPlugin.cc:727] Collision between 'bin8|assembly_battery_blue_4::link::collision' and 'kitting::vacuum_gripper_link::vacuum_gripper_link_collision'
[Dbg] [VacuumGripperPlugin.cc:774] Product attached to gripper: assembly_battery_blue named bin8|assembly_battery_blue_4
[Dbg] [VacuumGripperPlugin.cc:799] Drop scheduled
[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 155
[Dbg] [ROSAriacTaskManagerPlugin.cc:1266] Sim time: 156
[Dbg] [ROSVacuumGripperPlugin.cc:113] Gripper control requested: Disable
```

The last line is the kitting robot giving up when the planned trajectory gets off tolerance. Sometimes it takes a lot more time, with a lot more repetitions of the logs above, but the logs remain consistent.

The success and failed cases look mostly the same, except for the `[Dbg] [VacuumGripperPlugin.cc:806] Detaching product from gripper.` log in the failed case.

zeidk, 2021-05-30:
Can you please send me an email at zeid.kootbally@gmail.com? I will then reply with things I need from you. Thank you.

zeidk, 2021-06-01:
Please give it another try. It should be good now.

glpuga, 2021-06-02:
It seems to work now, thanks for looking into this!

closed 2021-06-02 by glpuga