usnistgov/ARIAC #112: kit_tray_x frames are not updated after dispatching an agv to a station

glpuga, 2021-05-25:
`kit_tray_x` frames are correctly set to the start position of the agv for any of the possible positions (ksN, asN) when the simulation starts.

However, when an agv is dispatched to an assembly station during the trial (e.g. `sample_multiple_non_high_priority_orders.yaml`) the corresponding `kit_tray_x` frame remains fixed at the initial pose instead of being updated to the destination pose of the agv.

zeidk, 2021-05-30:
Do you mean the transform returned for the kit tray after it has been moved to a station returns the wrong data?

glpuga, 2021-05-30:
Thanks for looking into this.

I mean the `kit_tray_x` transform relative to `world` remains fixed at the original position of the agv tray after the agv has been submitted to an assembly station, instead of being updated to the new pose of the agv/tray.

zeidk, 2021-05-30:
Ok, thanks. I'm looking into this now.

zeidk, 2021-05-31:
The latest push should fix this issue. Please test it and close the ticket if everything looks good. Thank you for reporting this issue.

glpuga, 2021-05-31:
Thanks, I'll check later tonight and update the ticket accordingly.

glpuga, 2021-06-01:
This is fixed. Thanks!

denbar-robotics, 2021-06-01:
weird, it is not fixed for me.  I am looking at the transform from world to kit_tray_2 for agv2, but the x coordinate is not moving when the agv moves, it just stays at -2.12

glpuga, 2021-06-01:
Reopening on request from @denbar-robotics .

I checked this on the latest master using  RViz and the trial configuration `multiple_non_high_priority_orders.yaml`.

After submitting AGV2, the `kit_tray_2` transform, as observed using RViz, would relocate to `as1`.

Also, some warnings my software logged before the update because of the parts on the tray becoming separated from the tray transform disappeared.

What trial configuration are you using to test this @denbar-robotics ?

denbar-robotics, 2021-06-02:
works now, thanks for re-opening, can close it.

closed 2021-06-02 by zeidk