usnistgov/ARIAC #113: New software release for finals breaks MoveIt! motion planning on kitting arm

mqt0029, 2021-05-29:
MoveIt! motion planning now throws `Invalid Trajectory: start point deviates from current robot state more than 0.01` for kitting arm elbow joint. All other trajectory seems to be working correctly.

glpuga, 2021-05-29:
Same for me.

The problem seems to be related to this change: https://github.com/usnistgov/ARIAC/blame/master/nist_gear/launch/kitting.launch#L29

Changing  `wrist_1_joint` to something in a smaller range (`-J wrist_1_joint 1.00`) fixed the problem for me.

mqt0029, 2021-05-29:
thanks for the suggestion @glpuga. This seems to be a sufficient temporary fix on my end.

glpuga, 2021-05-30:
yw!

Maybe it's better to leave this issue open so that it can be checked to see if it needs addressing on the competition code.

mqt0029, 2021-05-30:
Reopened in case this become an issue in final rounds.

zeidk, 2021-05-30:
Thank you. Yes, the moveit package was not modified (last modification was made 28 days ago). I will modify the launch file on my side and push it again.

glpuga, 2021-05-30:
This looks fixed for me now. Thanks!

closed 2021-05-30 by zeidk