usnistgov/ARIAC #114: flipped parts on assembly stations?

denbar-robotics, 2021-05-29:
There was an earlier note that flipped parts would not occur on assembly stations.  Specifically, it said that flipped parts may appear in bins, but that they should be correctly oriented during the kitting order and that there would be no flipped parts placed directly on the assembly stations.  However, the finals practice 2 config has a flipped pump on the assembly station.

denbar-robotics, 2021-05-29:
Found it, from the wiki:

Assembly
If the trial starts with AGVs already located at assembly stations, then no part flipping will be required for competitors. It is not an easy task to flip a part with the assembly robot while this challenge is much more manageable with the kitting robot.

Although part flipping is not present during assembly, competitors will still need to re-orientate the parts to place them in slots in the briefcases.

zeidk, 2021-05-30:
If the wiki said so then I will modify this :)

closed 2021-05-30 by zeidk