usnistgov/ARIAC #116: robot cannot pick the faulty part after the blocked sensor return normal !!!!!

YunpengHe, 2021-05-30:
We found a bug that:
      our robot can pick part normally, before and after sensor blocked. 
      when we pick the faulty part  (placed during sensor is blocked ) on agv, we make sure that the gripper of kitting robot is enabled. **And gripper is attached the faulty part obviously, but the gripper cannot pick up the part.** For other part, robot can pick normally !
[image]

YunpengHe, 2021-05-30:
For now, we have tested pump and sensor, these two part have the same issue as above

zeidk, 2021-05-30:
Do you get similar outputs as the ones described in #109 ?

YunpengHe, 2021-05-31:
yes, I have similar outputs

zeidk, 2021-06-01:
I'm working on this issue and it should be fixed today. However, the issue happens when the faulty gripper challenge is in use while your issue seems to be related to faulty sensors. Once the new push is made, give it a try and see if your issue is gone.

YunpengHe, 2021-06-02:
Yes, It works!

zeidk, 2021-06-02:
Great!!

closed 2021-06-02 by zeidk