usnistgov/ARIAC #118: There are is no sensor or regulator assembly task!!!

YunpengHe, 2021-06-01:
In the current final practices, there are no sensor or regulator assembly task, is it will evaluated in the final competition?

zeidk, 2021-06-01:
@Pavel-P This is related to the assembly plugin. We should give an answer today so that competitors are set on this matter.

denbar-robotics, 2021-06-01:
There was an earlier note that said there would only be two types of products for assembly tasks in the finals.  Please do not make a late change to that.

zeidk, 2021-06-01:
After discussion, the only parts used for assembly this year are batteries and pumps. I'm closing this ticket and I will convey this message in the next github push.

closed 2021-06-01 by zeidk

YunpengHe, 2021-06-02:
we have made a solution for assembly sensor or regulator......

zeidk, 2021-06-02:
I really appreciate the help. We won't include assembly with these 2 other parts as it requires more time for competitors to test new strategies. However, you can send me an email with your solution if you want. We would like to reuse the same scenarios and theme for ARIAC next year.