usnistgov/ARIAC #119: kit_tray_x transform is fixed

denbar-robotics, 2021-06-01:
Please reopen issue #112 as I am not seeing it fixed, although I pulled the latest update on master

I am looking at the transform from world to kit_tray_2 for agv2, but the x coordinate is not moving when the agv moves, it just stays at -2.12

[INFO] [1622576659.600593, 44.730000]: tray1 position x = -2.12
[INFO] [1622576659.601897, 44.731000]: tray2 position x = -2.12
[INFO] [1622576659.604087, 44.732000]: tray3 position x = -2.12
[INFO] [1622576659.605967, 44.733000]: tray4 position x = -2.12

glpuga, 2021-06-01:
Reopened.

zeidk, 2021-06-01:
Mmmm. I'll take a look.

zeidk, 2021-06-01:
Hi Dan, I just built a few kits on different AGVs and it seems to work for me.

I did not do a lookup transform in the code but in the terminal with `rosrun tf tf_echo world kit_tray_2`. The result is below. 
I'm not quite sure why it's not working for you. I'm about to do a new push in 1h or so. Try it once more when the code is pushed.

```
At time 196.050
- Translation: [-2.116, 1.368, 0.750]
- Rotation: in Quaternion [0.000, 0.000, -0.707, 0.707]
            in RPY (radian) [0.000, 0.000, -1.571]
            in RPY (degree) [0.000, 0.000, -90.000]
At time 197.050
- Translation: [-2.760, 1.368, 0.730]
- Rotation: in Quaternion [0.000, 0.000, -0.707, 0.707]
            in RPY (radian) [0.000, 0.000, -1.571]
            in RPY (degree) [0.000, 0.000, -90.000]
At time 198.050
- Translation: [-4.577, 1.368, 0.730]
- Rotation: in Quaternion [0.000, 0.000, -0.707, 0.707]
            in RPY (radian) [0.000, 0.000, -1.571]
            in RPY (degree) [0.000, 0.000, -90.000]
```

denbar-robotics, 2021-06-01:
OK

denbar-robotics, 2021-06-02:
works now :)

closed 2021-06-02 by denbar-robotics