usnistgov/ARIAC #121: new collisions

denbar-robotics, 2021-06-03:
With the new release, now getting internal collisions with the kitting arm:

[ WARN] [1622742615.631412014, 23.838000000]: Solution path may slightly touch on an invalid region of the state space
[ WARN] [1622742615.687259191, 23.862000000]: Solution path may slightly touch on an invalid region of the state space
[ INFO] [1622742615.688334569, 23.863000000]: SimpleSetup: Path simplification took 0.160338 seconds and changed from 10 to 4 states
[ERROR] [1622742615.693246985, 23.865000000]: Computed path is not valid. Invalid states at index locations: [ 40 41 42 61 62 63 64 65 ] out of 77. Explanations follow in command line. Contacts are published on /ariac/kitting/move_group/display_contacts
[ INFO] [1622742615.693356907, 23.865000000]: Found a contact between 'vacuum_gripper_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742615.693380304, 23.865000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1622742615.693518181, 23.865000000]: Found a contact between 'vacuum_gripper_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742615.693537399, 23.865000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1622742615.693634171, 23.865000000]: Found a contact between 'vacuum_gripper_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742615.693659107, 23.865000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1622742615.694133817, 23.865000000]: Found a contact between 'forearm_link' (type 'Robot link') and 'base_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742615.694157146, 23.865000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1622742615.694485811, 23.865000000]: Found a contact between 'wrist_1_link' (type 'Robot link') and 'upper_arm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742615.694504962, 23.865000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1622742615.695037307, 23.866000000]: Found a contact between 'wrist_2_link' (type 'Robot link') and 'upper_arm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742615.695062108, 23.866000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1622742615.695435304, 23.866000000]: Found a contact between 'wrist_2_link' (type 'Robot link') and 'upper_arm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742615.695455741, 23.866000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1622742615.696051033, 23.866000000]: Found a contact between 'vacuum_gripper_link' (type 'Robot link') and 'upper_arm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742615.696070141, 23.866000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ERROR] [1622742615.696241274, 23.866000000]: Completed listing of explanations for invalid states.
[ INFO] [1622742615.696376392, 23.866000000]: Planning attempt 2 of at most 5
[ INFO] [1622742615.697456689, 23.867000000]: Planner configuration 'kitting_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1622742615.697624047, 23.867000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1622742616.937489056, 24.363000000]: Received event 'stop'
[ INFO] [1622742617.108299686, 24.426000000]: RRTConnect: Created 1426 states (939 start + 487 goal)
[ INFO] [1622742617.108348282, 24.426000000]: Solution found in 1.410778 seconds
[ WARN] [1622742617.210564787, 24.462000000]: Solution path may slightly touch on an invalid region of the state space
[ WARN] [1622742617.278919485, 24.488000000]: Solution path may slightly touch on an invalid region of the state space
[ INFO] [1622742617.280892964, 24.489000000]: SimpleSetup: Path simplification took 0.172494 seconds and changed from 13 to 4 states
[ERROR] [1622742617.289422582, 24.492000000]: Computed path is not valid. Invalid states at index locations: [ 77 78 79 80 81 82 ] out of 92. Explanations follow in command line. Contacts are published on /ariac/kitting/move_group/display_contacts
[ INFO] [1622742617.289504742, 24.492000000]: Found a contact between 'vacuum_gripper_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1622742617.289522455, 24.492000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)

denbar-robotics, 2021-06-03:
Here is an screenshot when the internal kitting arm collision occurs.   Happens every time at this spot, but picks up OK at most spots.
[image]

zeidk, 2021-06-03:
OK, I'll see what I can do.

denbar-robotics, 2021-06-03:
It's  somehow related to long Cartesian moves.  I got rid of it by making the maximum single move length = 5.0.
No rush to fix it before the finals unless others encounter it as well.

zeidk, 2022-04-11:
Closing this since it was related to ARIAC 2021. Reopen if the issue persists.

closed 2022-04-11 by zeidk