usnistgov/ARIAC #124: ARIAC 2022 platform clarification

130s, 2022-01-11:
I can't find anywhere, but I _think_ I overheard in the kickoff meeting that the 2022 competition will be on Ubuntu 18.04 + ROS Melodic. Is that true? If so, I'd be curious to hear the rationale to stick to it.

I'm personally concerned particularly with ROS Melodic. My gut feeling was/is that why not move on to 20.04 + Noetic.
- Particularly, [Melodic officially uses Python version 2 (www.ros.org/rep-0003)](https://www.ros.org/reps/rep-0003.html#melodic-morenia-may-2018-may-2023), meaning a lot of tools have to stick to Python2 (or you need to circumvent by yourself to use Py3, which maybe non-trivial unless you're skillful or experienced. But IMO Py3 migration can be a huge pain). While Python might not be the core language for most of competitors' programs, this limits miscellaneous tools (and some teams may rely on Python a lot integrating AI stuff etc. that are only available in Python?). Python community ended Py2 support a few years ago and lots of good new development has been made only in Py3.
- I can't find 2021 ROS user demography (asked [answers.ros.org#394076](https://answers.ros.org/question/394076/2021-metrics-missing/)) but I can only assume there's no issue stability-wise in either Melodic or Noetic.
- I can tell by using Ubuntu 18.04 previous participants can use the same platform w/o change. I can't speak for how much priority that can be. But I can tell new participants will be irritated.

AravindaDP, 2022-01-13:
Was there a kickoff meeting for 2022? Or are you referring to last year workshop meeting on October?

I don't see any new announcements on https://www.nist.gov/el/intelligent-systems-division-73500/agile-robotics-industrial-automation-competition nor did not received any updates even though we registered last month through https://bit.ly/ariac2022

130s, 2022-01-13:
> Was there a kickoff meeting for 2022? Or are you referring to last year workshop meeting on October?

That one. Sorry I'm not so sure if the platform announcement I'm talking about was for 2022, or for 2021 (hence "clarification" in the title of this ticket).

zeidk, 2022-01-25:
We wanted to focus on the addition of new challenges in the 2022 competition while reusing the same platform from previous year. On top of adding new challenges, we addressed some issues we had in the previous version (such as the assembly plugin, activating/deactivating robot controllers on the fly). However, I fully understand that most Linux users have migrated to noetic and going back to melodic may be a hassle. We will switch platform for next year competition.

closed 2022-01-25 by zeidk

130s, 2022-01-25:
Thanks, understood. I just want to bring this up in a slim hope there'd be a room to change the platform to newer distro.

Good thing I later found is that Ubuntu 18.04 might be still the majority at least among ROS users, interpreting from https://metrics.ros.org/packages_rosdistro.html
[image]