usnistgov/ARIAC #127: motion planning fails for the full group

denbar-robotics, 2022-03-29:
Moveit robot planning in the assembly robot works for the gantry_torso and for the gantry_arm groups but it fails for the combined gantry_full group.  

It's a pretty simple fix to gantry_moveit_config/config/kinematics.yaml  Just make the solver for gantry_full the same as the solver for the other two groups.  Doing that makes gantry_full planning work fine.
I found that someone else previously ran into the same bug and has the [same solution](https://githubhot.com/repo/usnistgov/ARIAC/issues/108).

- kinematics_solver: cached_ik_kinematics_plugin/CachedSrvKinematicsPlugin
+ kinematics_solver: kdl_kinematics_plugin/KDLKinematicsPlugin

zeidk, 2022-03-29:
I will take a look and see how it behaves. What happens when it fails? What do the errors look like?

denbar-robotics, 2022-03-30:
(plan, fraction) = group.compute_cartesian_path(
    waypoints,  # waypoints to follow
    0.01,  # eef_step
    0.0,  # jump_threshold = max radians allowed in a joint for a single step
    False)  # do not avoid collisions

[ERROR] [1645207759.372480196, 533.557000000]: Seed state must have size 9 instead of size 13
[ERROR] [1645207759.373198667, 533.557000000]: Service call failed to connect to service: /ariac/gantry/solve_ik

zeidk, 2022-03-31:
It should be pushed now.

closed 2022-03-31 by zeidk

denbar-robotics, 2022-04-01:
It works fine, thanks.