usnistgov/ARIAC #128: agv at asm1 and trays table conflict

denbar-robotics, 2022-03-29:
Will a tray ever be needed when AGV1 is at ASM1?  The AGV post makes it tough to get access to the entire table if there is a camera placed above the table.  However, for that to happen, there would have to be a kitting order issued at the same time as an ongoing assembly order at ASM1.  Is that excluded?

zeidk, 2022-03-30:
This situation is possible. You may receive a regular order (order_1 with a kitting shipment) after the kitting shipment from order_0 is submitted to AS1. There are a few things I can do to make life easier. 
- Make sure to not use AS1 in order_0 if order_1 requires kitting.
- Make sure there is no kitting in order_1 if there is an AGV at AS1.
- Get rid of one of the two table trays and give the remaining table the ability to have different movable tray models.

The last one requires a bit more work but in the long run it will make my life easier when designing trial for the qualifiers and finals.

denbar-robotics, 2022-03-30:
Any/All of those sound good.

denbar-robotics, 2022-03-30:
Turns out I was wrong on this.  By changing the approach direction, the table is fully reachable even with the AGV in place and a camera above.
Sorry for bothering you with this.

zeidk, 2022-04-02:
It's ok, we don't actually need two tables for the trays. I made some changes about this. When you get a chance, can you try the "ariac2022_movable_trays" branch. Check the updates page first. I also updated the scoring section to answer your previous question.

denbar-robotics, 2022-04-02:
Happy to take a look at the new tables, but as I mentioned above, there is no need for a change.  The tables are fully accessible as they are in the original setup.

I see in the new branch the scoring clarification that a movable tray in the wrong pose eliminates the bonus for that shipment but does not otherwise impact the score.  Thanks for making that clear.

denbar-robotics, 2022-04-03:
The "ariac2022_movable_trays" branch fails on startup with the message:
Error: unknown top level entry 'table_tray_infos'

zeidk, 2022-04-04:
Sounds good

denbar-robotics, 2022-04-04:
resolved

closed 2022-04-04 by denbar-robotics