usnistgov/ARIAC #129: part placement

denbar-robotics, 2022-03-30:
Is the ordered part placement location and orientation with respect to the AGV frame or the movable tray frame that is on the AGV?
Seems like If the tray is slightly misplaced and the placement location is with respect to the AGV frame, then correctly placed parts may end up on a movable tray frame edge.  Is that correct?

zeidk, 2022-03-30:
Each product pose described in an order is relative to the movable tray. The contact sensor of the movable tray will report parts in the movable tray and the pose of the movable tray relative to the static kit tray (e.g., kit_tray_1).

denbar-robotics, 2022-03-30:
thanks for the clarification.

denbar-robotics, 2022-03-30:
What is the impact on the score if the correct movable tray is placed in the wrong pose, but the parts are all correct in the movable tray frame?  The scoring metrics do not address the issue of the movable tray pose being wrong.

zeidk, 2022-04-11:
The wiki addresses this question. See section [Scoring](https://github.com/usnistgov/ARIAC/blob/ariac2022/wiki/documentation/scoring.md#movable-tray)

closed 2022-04-11 by zeidk