usnistgov/ARIAC #133: Questions regarding movable trays

glpuga, 2022-04-04:
Are these scenarios possible?

- The movable tray needs to be replaced for one of a different type, either because the scene started with an unwanted movable tray in the agv (I think this is not possible to configure in the yaml file) or because of an update the a previous order (this case is probably possible). This would require removing all the parts, replacing the tray, then loading the new parts on.
- An agv used for a kitting order starts the simulation with parts loaded on them but no movable tray, requiring all parts to be removed  to place the movable tray before completing the rest of the order.

zeidk, 2022-04-04:
Hello, none of these scenarios will be encountered in a trial. Thanks for the questions, I will update the wiki with them.

glpuga, 2022-04-04:
Thanks for the answer!

zeidk, 2022-04-11:
Wiki updated to answer your questions

closed 2022-04-11 by zeidk