usnistgov/ARIAC #134: trays move around on agvs

denbar-robotics, 2022-04-04:
The movable tray slips and slides around very easily on the agv.  Once it moves, it can be very hard to reset into place, as there may be parts sitting on it.  Is there any way to increase the friction or perhaps lock the tray in place on the agv?

zeidk, 2022-04-04:
Yes, we can lock it in place through a service call. The code for this service is already in an ARIAC .cc file, I just need to uncomment it.

denbar-robotics, 2022-04-04:
Thanks.  And appreciate the response late on a Sunday night!

zeidk, 2022-04-04:
I will make another push tomorrow, which will also address the other questions from other users.

:)

zeidk, 2022-04-11:
Did you get a chance to try the service which locks/unlocks movable trays? Can I close this ticket?

denbar-robotics, 2022-04-11:
I didn't see that service show up.  I'll look for it and try tonight.

zeidk, 2022-04-11:
Ok, see the [updates](https://github.com/usnistgov/ARIAC/blob/ariac2022/wiki/misc/updates.md#release-2022-april-9) for more information.

denbar-robotics, 2022-04-12:
The tray stays in place now but the parts go flying off it when submitted.

zeidk, 2022-04-12:
Well I tried this situation a few times and it worked fine. I'll check what's wrong.

denbar-robotics, 2022-04-12:
When calling the service from the command line, the returned result is False, but the message says it worked.  And it does work.  Also, you get messages that say the tray is locked/unlocked from agvs that have no movable trays on them

rosservice call /ariac/kit_tray_2/lock
success: False
message: "Movable tray is now locked on the kit tray"

Also, it works from the command line (as shown above) but fails when called within a python script.  wait_for_service works, so the service name is right, but the ServiceProxy call generates this exception:

Service call failed:  'ServiceProxy' object has no attribute 'success'