usnistgov/ARIAC #135: startup crash

denbar-robotics, 2022-04-05:
About once every five or so launches, gear fails with this message:

[ INFO] [1649198019.310798403]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ WARN] [1649198019.315425540]: spawnURDFModel service called
[ WARN] [1649198019.316320590]: spawnSDFModel service called
[INFO] [1649198019.349839, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1649198019.423852, 0.000000]: Calling service /gazebo/spawn_urdf_model
gzserver: /usr/include/boost/smart_ptr/enable_shared_from_this.hpp:51: boost::shared_ptr<X> boost::enable_shared_from_this<T>::shared_from_this() [with T = gazebo::transport::Node]: Assertion `p.get() == this' failed.
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
[ERROR] [1649198021.671248, 0.000000]: Spawn service failed. Exiting.
[ERROR] [1649198021.671190, 0.000000]: Spawn service failed. Exiting.
[ERROR] [1649198021.671297, 0.000000]: Spawn service failed. Exiting.
Aborted

zeidk, 2022-04-05:
I will look into this today. It has to do with one of the Gazebo plugins.

zeidk, 2022-04-06:
Could you send me your performance.log file whenever this issue arises again. I cannot reproduce it on my end but it seems a plugin is not loading due to a slow machine.

denbar-robotics, 2022-04-08:
sent

zeidk, 2022-05-27:
I haven't see this issue for a while. Reopen if you still see it.

closed 2022-05-27 by zeidk

db7296, 2022-05-27:
Yes, it still happens.  There is no option available to re-open.  Should I create a new issue?