usnistgov/ARIAC #136: Sample environment crashes

prasuchit, 2022-04-06:
Hi,

This is my first time participating in the competition and I was following the docs to setup the environment. Following [this](https://github.com/usnistgov/ARIAC/blob/ariac2022/wiki/tutorials/gear_interface.md#start-gear), when I ran the sample environment, it keeps crashing. This are the issues I see:
**Warnings:

Warning [parser.cc:950] XML Element[activation_topic], child of element[camera] not defined in SDF. Ignoring[activation_topic]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[activation_topic], child of element[camera] not defined in SDF. Ignoring[activation_topic]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.

[WARN] [1649280325.118710, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[ariac/gantry/gantry_controller_spawner-5] process has finished cleanly
log file: /home/prasanth/.ros/log/fe86749a-b5ef-11ec-8bcc-54bf647decd4/ariac-gantry-gantry_controller_spawner-5*.log
[WARN] [1649280325.398539, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1649280325.420090, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1649280325.473957, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1649280325.502658, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1649280325.608992, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.

**Errors:
[ARIAC_ERRORS.txt](https://github.com/usnistgov/ARIAC/files/8431373/ARIAC_ERRORS.txt)

I've made sure my ROS installation is proper and that I have all the necessary controllers, etc installed. Kindly help.

glpuga, 2022-04-07:
Hi,

I think I needed t add these to get the simulation and the demos up and running:

```
    "ros-melodic-ros-control*"
    "ros-melodic-control*
    "ros-melodic-gazebo-ros-control*"
    ros-melodic-moveit-simple-controller-manager 
    ros-melodic-moveit-commander 
 ```
 
 I think you may be missing the fourth one.

prasuchit, 2022-04-07:
Hi @glpuga,

I just installed all the packages you mentioned and you were right that I was missing a couple of them. However, that didn't fix the problem, I still see the same warnings and errors and gazebo crashes.

I think the launch file is missing the controller spawner. Maybe something like this?

`

    <node name="controller_spawner"
      pkg="controller_manager"
      type="spawner" respawn="false"
      output="screen" ns="/nist_gear"
      args="joint_state_controller
        FIRST_effort_controller
        SECOND_effort_controller
        THIRD_effort_controller
        FOURTH_effort_controller"
    />`

rmerriam, 2022-04-07:
I am seeing the failure of the sample to load. I have done Ariac previously. 

1. The directions for installation are missing the installation of the Movit controllers. I installed them. Failure may have changed but still exists.
2. Gazebo version is 9.19 but the installation direction says 9.16. 

Some errors during start:
`
Error [parser_urdf.cc:3167] Unable to call parseURDF on robot model
Error [parser.cc:406] parse as old deprecated model file failed.
Error [parser_urdf.cc:3167] Unable to call parseURDF on robot model
Error [parser.cc:406] parse as old deprecated model file failed.
Error [parser_urdf.cc:3167] Unable to call parseURDF on robot model
Error [parser.cc:406] parse as old deprecated model file failed.
<< later >>
[ERROR] [1649354269.605302543, 1.246000000]: Could not stop controller 'static_linear_arm_actuator_controller' since it is not running
[ERROR] [1649354269.756013292, 1.248000000]: Could not stop controller 'static_shoulder_pan_joint_controller' since it is not running
`

zeidk, 2022-04-08:
Gazebo 9.19 is fine as this is the version I am running.
The controllers are loaded from kitting.launch and gantry.launch.
The Errors [parser_urdf.cc:3167] and parser.cc:406] are expected and will not affect the software.

Can you try to increase the [unpause time](https://github.com/usnistgov/ARIAC/blob/7b2341d6f9cae7c3ebd48827c55b5ddb990352a3/nist_gear/script/gazebo_unpauser.py#L33)? Try 15 and 20. It looks like Gazebo is unpausing before the controllers were properly initialized. This happens with slow systems.

rmerriam, 2022-04-08:
> Can you try to increase the [unpause time](https://github.com/usnistgov/ARIAC/blob/7b2341d6f9cae7c3ebd48827c55b5ddb990352a3/nist_gear/script/gazebo_unpauser.py#L33)? Try 15 and 20. It looks like Gazebo is unpausing before the controllers were properly initialized. This happens with slow systems.

I increased to 20 and then to 40 but Gazebo still stopped. My system is not particularly slow. After the following lines there is a long pause and then the 'Killed' line. 

```
[INFO] [1649458929.609707, 0.061000]: Controller Spawner: Loaded controllers: static_linear_arm_actuator_controller, static_shoulder_pan_joint_controller, static_shoulder_lift_joint_controller, static_elbow_joint_controller, static_wrist_1_controller, static_wrist_2_controller, static_wrist_3_controller
[INFO] [1649458929.754757, 0.065000]: Controller Spawner: Loaded controllers: static_shoulder_pan_joint_controller, static_shoulder_lift_joint_controller, static_elbow_joint_controller, static_wrist_1_controller, static_wrist_2_controller, static_wrist_3_controller
<<<  pause >>>
Killed
```

I'm not seeing any messages from the "unpause" python code in the logs or output.

rmerriam, 2022-04-15:
> Do you mind going through the [Installation](https://github.com/usnistgov/ARIAC/blob/ariac2022/wiki/tutorials/installation.md) page one more time?

You can close this for me. I'm likely passing on the competition.

prasuchit, 2022-04-15:
You can go ahead and close this issue, I was able to get it working within a docker. Gazebo loads fully and I see all the models, topics and services. However, I still see xacro/urdf parsing errors, you might want to check that out:

Error [parser_urdf.cc:3167] Unable to call parseURDF on robot model
Error [parser.cc:406] parse as old deprecated model file failed.

closed 2022-04-16 by zeidk