usnistgov/ARIAC #137: sensor falls out of briefcase

denbar-robotics, 2022-04-09:
I am wondering if anyone else has this issue:  When the sensor is placed in the briefcase and released, it just falls out onto the floor.

zeidk, 2022-04-11:
@jaybrecht @Pavel-P Could you please assist with this ticket? It seems an advice is required here instead of an actual technical fix.

jaybrecht, 2022-04-11:
Here is a video of the test competitor installing the sensor to briefcase_1. 

https://user-images.githubusercontent.com/50928580/162780655-70c6518a-e6a6-4ced-b0d9-a10f74efdecd.mp4

You can test this yourself. In the first terminal run
`roslaunch test_competitor assembly.launch station:=as1`
In the second run:
`rosrun test_competitor assembly_commander_node.py`

denbar-robotics, 2022-04-11:
oh, didn't know that. I'll give it a try.  Is that true for the regulator too?

jaybrecht, 2022-04-11:
Yes the regulator works in a similar way. I will close this issue for now but let us know if you have any other issues.

closed 2022-04-11 by jaybrecht