usnistgov/ARIAC #138: pose score

denbar-robotics, 2022-04-10:
I think I have the regulator and battery both properly placed, but the scoring only gives credit for one pose.  I cannot figure out which pose is wrong and why.

Here is the score:
      ...number of products shipped with correct type (CT): [2]
      ...number of products shipped with correct pose (CP): [1]
      ...number of products shipped with correct color (CC): [2]

Here are the locations of the parts:

      type: "assembly_regulator_red"
      is_faulty: False
      pose: 
        position: 
          x: -0.216139300983
          y: -0.165686622478
          z: 0.209974810976
        orientation: 
          x: 0.0179735639226
          y: 0.705447186881
          z: -0.00631499908424
          w: 0.708506413738
    - 
      type: "assembly_battery_green"
      is_faulty: False
      pose: 
        position: 
          x: -0.0268073019699
          y: 0.178264655203
          z: 0.0880191920094
        orientation: 
          x: -0.00523549024672
          y: -0.00336591763012
          z: 0.00354695380635
          w: 0.99997433935

denbar-robotics, 2022-04-11:
The above was the output from get_content.  I think the answer is related to the question of why the sensor falls out-- I think it may be because I did not push down hard enough on the regulator

denbar-robotics, 2022-04-11:
I'll check that and, if it is not the solution, will re-open.

denbar-robotics, 2022-04-15:
When running your sample code, it properly places all 4 parts into the briefcase, but the score is wrong:
Also, get_contents is right while the run is going, but after the run, with all 4 parts properly placed, only the regulator shows up.  I am seeing similar things with my runs.  By the way, is there any way to know if a sensor or regulator is locked in?  Is there any way to unlock it in the case where its pose is wrong and you want to change it?

roslaunch test_competitor assembly.launch station:=as1
rosrun test_competitor assembly_commander_node.py

 rosservice call /ariac/briefcase_1/get_content
shipment: 
  briefcase_id: "station1::briefcase_1"
  products: 
    - 
      type: "assembly_regulator_red"
      is_faulty: False
      pose: 
        position: 
          x: -0.218242592306
          y: -0.171336389141
          z: 0.210253423153
        orientation: 
          x: 0.00894504378811
          y: 0.703750238344
          z: 0.0194790056573
          w: 0.710124043081

   Order Score
------------------------
   ...order ID [order_0]
   ...total order score: [3]
   ...completion score: [3]
   ...time taken: [153.465]
   ...priority: [1]
------------------------
      Assembly Score
------------------------
      ...shipment type: [order_0_assembly_shipment_0]
      ...completion score: [3]
         +2pts x (nb of products with CT & CP): [2]
           +1pt x (nb of products with CC): [1]
             +bonus: 4pts x (nb of products with CT & CP & CC) = [0]
      ...complete: [false]
      ...shipped: [true]
      ...number of products shipped: [2]
      ...has faulty products: [false]
      ...has missing products: [true]
      ...has unwanted products: [false]
      ...number of products shipped with correct type (CT): [2]
      ...number of products shipped with correct pose (CP): [1]
      ...number of products shipped with correct color (CC): [2]
      ...correct assembly station: [true]

zeidk, 2022-04-18:
Got it, we should get this fixed tomorrow.

zeidk, 2022-05-01:
There is a service that was created to get parts fully connected to a briefcase. This is a way to know if a part is not properly inserted. The service can be run in competition mode. Check the updates page on the wiki if you missed it. 

In other words, if a part does not show up on this service call it means the part was not properly inserted and you should regrasp the part and re-insert it.

I'm closing this for now.

closed 2022-05-01 by zeidk