usnistgov/ARIAC #139: submitted agv1 goes to a different location than spawned as1

denbar-robotics, 2022-04-11:
When a kitting shipment is submitted to as1, the AGV moves in X beyond where as1 is located when an AGV is spawned there.

zeidk, 2022-04-11:
What if I make the AGV spawn x position the same as the x position when it reaches an assembly station after it has been shipped? Will this work or should it be the other way around? The first solution is faster for me.

denbar-robotics, 2022-04-11:
I think that changing the spawn location would probably be OK if it means the fix could be in quickly.  However, it is definitely more work on this end, since the current code is based on the original spawn point.  Hard to know if it would cause issues with something like overlapping camera coverage.  Would have to test to be sure.

zeidk, 2022-04-11:
I can make the AGV move to the same spawn location. I looked at the code to do that and in the end it's not much work. Hopefully that will not break other competitors' code. I will wait 1h or 2 to see if anyone is against this decision.

zeidk, 2022-04-11:
Should be fixed now.

closed 2022-04-11 by zeidk