usnistgov/ARIAC #140: seg faults

denbar-robotics, 2022-04-11:
I am encountering segmentation faults in various situations.  An example log sent.  Mostly they happen when both kitting and gantry orders exist.

zeidk, 2022-04-11:
Does it happen when both kitting and assembly are in the same order? When in separate orders? Or both? Which YAML config file did you use?

denbar-robotics, 2022-04-11:
The most consistent one I see is with kitting orders-- the gantry robot places the trays and the seg fault happens when the kitting arm tries to pick up a part from bin1 to place on a tray that is sitting on agv1.  The one here was with a single order that had both kitting and assembly and it occured when picking up an assembly part on agv2 at as1 after delivering a tray to agv1 at ks1.
Do you mean which trial config yaml?  I don't recall, but I can note that next time it happens.

denbar-robotics, 2022-04-16:
Here is the one that consistently occurs when moving the kitting arm to agv1 when there is a movable tray there.  Log files sent.

Segmentation fault
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 16965, exit code 139, cmd /home/dbarry/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode -r --record_period 0.01 --record_path=~/.ariac/log/gazebo /tmp/ariac/ariac.world __name:=gazebo __log:=~/.ros/log/99de1de6-bd4b-11ec-b1b4-181deaec5b6c/gazebo-2.log].
log file: ~/.ros/log/99de1de6-bd4b-11ec-b1b4-181deaec5b6c/gazebo-2*.log
Initiating shutdown!
================================================================================

zeidk, 2022-04-18:
I will email you about this issue.

liuhaoran1, 2022-04-20:
> I will email you about this issue.

I have also encountered this problem, how to solve it？

liuhaoran1, 2022-04-20:
> I am encountering segmentation faults in various situations. An example log sent. Mostly they happen when both kitting and gantry orders exist.
> 
> Here is one example of console output (sorry about the bold and the large font size-- don't know what generates that).
> # [ INFO] [1649660551.542935522, 67.966000000]: Controller /ariac/kitting/kitting_arm_controller successfully finished
> [ INFO] [1649660551.543034495, 67.966000000]: Completed trajectory execution with status PREEMPTED ...
> Segmentation fault
> ================================================================================REQUIRED process [gazebo-2] has died!
> process has died [pid 15035, exit code 139, cmd ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode -r --record_period 0.01 --record_path=.ariac/log/gazebo /tmp/ariac/ariac.world __name:=gazebo __log:=.ros/log/dde87ada-b964-11ec-b40e-181deaec5b6c/gazebo-2.log].
> log file: .ros/log/dde87ada-b964-11ec-b40e-181deaec5b6c/gazebo-2*.log
> Initiating shutdown!

Have you solved the problem yet?I have the same problem.

db7296, 2022-05-01:
Still have seg faults with the latest update.  This was during a combined kitting/assembly trial.  Logs sent.

[ INFO] [1651364112.826921049, 233.164000000]: Computed Cartesian path with 33 points (followed 100.000000% of requested trajectory)
[ WARN] [1651364112.832428606, 233.165000000]: Dropping first 1 trajectory point(s) out of 33, as they occur before the current time.
First valid point will be reached in 0.628s.
Segmentation fault
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 29952, exit code 139, cmd /home/dbarry/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode -r --record_period 0.01 --record_path=/home/dbarry/.ariac/log/gazebo /tmp/ariac/ariac.world __name:=gazebo __log:=/home/dbarry/.ros/log/a7a4b31c-c8e1-11ec-893a-181deaec5b6c/gazebo-2.log].
log file: /home/dbarry/.ros/log/a7a4b31c-c8e1-11ec-893a-181deaec5b6c/gazebo-2*.log
Initiating shutdown!
================================================================================

db7296, 2022-05-01:
Looks like it happens when submitting and making the agv move:

[ INFO] [1651436626.459820428, 94.305000000]: [SUCCESS] [agv1: ks1->as1 triggered]
Segmentation fault
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 28832, exit code 139, cmd ~/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode -r --record_period 0.01 --record_path=~/.ariac/log/gazebo /tmp/ariac/ariac.world __name:=gazebo __log:=~/.ros/log/ab7830e0-c98b-11ec-a206-181deaec5b6c/gazebo-2.log].
log file: ~/.ros/log/ab7830e0-c98b-11ec-a206-181deaec5b6c/gazebo-2*.log
Initiating shutdown!
================================================================================
[ariac/kitting/move_group-17] killing on exit

zeidk, 2022-05-06:
Yep, found the source of the problem when kitting and assembly are both in a trial with a high-priority order in the trial. I haven't checked with having kitting and assembly as two regular orders (assembly announced after agv is shipped). Do you also have this issue with two regular orders?

db7296, 2022-05-06:
Yes, still seg faults, although I have not checked whether it happens at that exact spot.  Here is one with just the kitting arm moving, no assembly activity at all.  Two orders, the first one finished and the second one is in progress.  Logs to follow.

I'm curious what caused the one you solved?

[ WARN] [1651815447.043326733, 65.785000000]: Dropping first 1 trajectory point(s) out of 6, as they occur before the current time.
First valid point will be reached in 0.914s.
Segmentation fault
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 7872, exit code 139, cmd /home/dbarry/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode -r --record_period 0.01 --record_path=/home/dbarry/.ariac/log/gazebo /tmp/ariac/ariac.world __name:=gazebo __log:=/home/dbarry/.ros/log/ebbf3f52-ccfd-11ec-b701-181deaec5b6c/gazebo-2.log].
log file: /home/dbarry/.ros/log/ebbf3f52-ccfd-11ec-b701-181deaec5b6c/gazebo-2*.log
Initiating shutdown!
================================================================================

db7296, 2022-05-06:
The answer to your question is in the comment above, but here is an error I saw during a docker run with kitting.yaml.  It may not be related to the seg faults, but referencing None objects can produce memory leaks that lead to them:

xception in thread /ariac/kitting/arm/gripper/state:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_pubsub.py", line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 846, in receive_loop
    self.close()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'

liuhaoran1, 2022-05-07:
The problem is still puzzling me .

db7296, 2022-05-07:
Have you run a profiler on the code?  Most seg faults I run into are either a memory leak or trying to delete a null pointer.

denbar-robotics, 2022-05-31:
fixed.

closed 2022-05-31 by denbar-robotics