usnistgov/ARIAC #143: Gripper fault: the dropped part hit other parts and the movable tray.

barrygxwan, 2022-04-15:
Is there a possible that: when robot do kitting task, the part drop into AGV tray due to the gripper faulty, the dropped part will hit other parts that have placed or the dropped part hit the movable tray result in tray moving?

AravindaDP, 2022-04-16:
Also does this trigger happens only once? (i.e. It is possible to Pick the part and reattempt placement assuming part wasn't fallen on the floor) Or it essentially need to use the gantry robot (if this happens in kitting area with kitting robot) or skip with the part since reattempt will most certainly trigger the drop part event again

zeidk, 2022-04-18:
@AravindaDP This should happen only once. You can regrasp this part.
@barrygxwan The dropped part challenge actually teleports the part and does not really drop it. It may collide with other parts but it is unlikely to affect the movable tray. You can also use the lock/unlock service on the movable tray if you are afraid the movable tray may move.

zeidk, 2022-04-18:
I will add this to the wiki.

closed 2022-04-18 by zeidk