usnistgov/ARIAC #148: submit service fails

dan9thsense, 2022-04-19:
The submit service often fails with this message and it crashes the sim:

terminate called after throwing an instance of 'std::out_of_range'
[ERROR] [1650398217.055792155, 264.546000000]: Exception thrown while processing service call: basic_string::substr: __pos (which is 19) > this->size() (which is 18)
  what():  basic_string::substr: __pos (which is 19) > this->size() (which is 18)

liuhaoran1, 2022-04-20:
This was also the case when I submitted the Assembly

mqt0029, 2022-05-02:
Now seeing the same issue. It seems enabling the `--development-flag` fixes a lot of things...

AravindaDP, 2022-05-06:
I also get the same error with `roslaunch test_competitor assembly.launch station:=as1` and assembly_commander example

It does not fail with `roslaunch test_competitor assembly.launch station:=as1_simple` but Score is 0.

db7296, 2022-05-31:
this is now fixed.