usnistgov/ARIAC #155: bug no solution

liuhaoran1, 2022-04-28:
Now, in the development process, I encountered the following problem, do not know how to solve:

1.No score after assembly regulator

    When I use this command:rosservice call /ariac/briefcase_2/get_content，it does not display regulator's information，but I'm sure it  put  in briefcase_2.

2.No bonus 3x points for high priority orders

   I submit the high priority order first and then work on the low priority order

3.agv2.3.4 shipment product drops（no solution）

   When I submit the kitting task on AGV2,3 and 4, the products and movable trays will fall to the ground during transportation, but this will not happen in AGV1. The final score display does not affect the score situation.

4.robot failure problem（no solution）

  When I design the robot to fail in the order, the whole interface stops and time stands still

zeidk, 2022-04-29:
For 1. If the regular is locked on the side of the briefcase then it should have been reported by the service call. I will take a look right now and make the fix.
For 2. What was the first order that was submitted? A kitting order or assembly? Do you mind posting the YAML config file you used?
For 3. This has been fixed in the current version which will be pushed tonight.
For 4. This has been fixed.

liuhaoran1, 2022-04-29:
# ARIAC 2022
# Trial where a simple kit is required.
# The kit must be built on a tray of type movable_tray_metal_shiny
# The pose of the parts are in the movable tray frame.
# Once the kit is built, it must be shipped to as1

options:
  insert_models_over_bins: true
  belt_population_cycles: 0
  insert_models_over_stations: false
  gazebo_state_logging: false
  # mandatory: gripper_tray or gripper_part
  current_gripper_type: gripper_tray
time_limit: 500

# The table_tray_infos is not mandatory
# The different options for tray_model are:
#  -movable_tray_metal_shiny
#  -movable_tray_metal_rusty
#  -movable_tray_dark_wood
#  -movable_tray_light_wood
# quantity can be: 1, 2, or 3
table_tray_infos:
  table_1: 
    tray_model: movable_tray_metal_shiny
    quantity: 1
  table_2: 
    tray_model: movable_tray_dark_wood
    quantity: 1

# mandatory: location of agvs and products at start
agv_infos:
  agv1:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0, 0]
          rpy: [0, 0, 'pi/4']
  agv2:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0.2, 0]
          rpy: [0, 0, 'pi/4']
  agv3:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0, 0]
          rpy: [0, 0, 0]
  agv4:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0.1, 0]
          rpy: [0, 0, 0] 

# Only one order 
orders:
  order_0:
    priority: 1
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: time
    announcement_condition_value: 0.0
    assembly:
      shipment_count: 1
      stations: [as4]
      products:
        part_0:
          type: assembly_battery_red
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 'pi']
        part_1:
          type: assembly_pump_red
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]
  order_1:
    priority: 3
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: unwanted_products
    announcement_condition_value: 1
    assembly:
      shipment_count: 1
      stations: [as1]
      products:
        part_0:
          type: assembly_battery_blue
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 0]
        part_1:
          type: assembly_pump_blue
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]

models_over_bins:
   bin2:
     models:
       assembly_pump_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
   bin6:
     models:
       assembly_battery_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
faulty_products:
  - assembly_pump_green_1
  - assembly_battery_green_2

# don't forget to include at least 1 empty line here

liuhaoran1, 2022-04-29:
`# ARIAC 2022
# Trial where a simple kit is required.
# The kit must be built on a tray of type movable_tray_metal_shiny
# The pose of the parts are in the movable tray frame.
# Once the kit is built, it must be shipped to as1

options:
  insert_models_over_bins: true
  belt_population_cycles: 0
  insert_models_over_stations: false
  gazebo_state_logging: false
  # mandatory: gripper_tray or gripper_part
  current_gripper_type: gripper_tray
time_limit: 500

# The table_tray_infos is not mandatory
# The different options for tray_model are:
#  -movable_tray_metal_shiny
#  -movable_tray_metal_rusty
#  -movable_tray_dark_wood
#  -movable_tray_light_wood
# quantity can be: 1, 2, or 3
table_tray_infos:
  table_1: 
    tray_model: movable_tray_metal_shiny
    quantity: 1
  table_2: 
    tray_model: movable_tray_dark_wood
    quantity: 1

# mandatory: location of agvs and products at start
agv_infos:
  agv1:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0, 0]
          rpy: [0, 0, 'pi/4']
  agv2:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0.2, 0]
          rpy: [0, 0, 'pi/4']
  agv3:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0, 0]
          rpy: [0, 0, 0]
  agv4:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0.1, 0]
          rpy: [0, 0, 0] 

# Only one order 
orders:
  order_0:
    priority: 1
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: time
    announcement_condition_value: 0.0
    assembly:
      shipment_count: 1
      stations: [as4]
      products:
        part_0:
          type: assembly_battery_red
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 'pi']
        part_1:
          type: assembly_pump_red
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]
  order_1:
    priority: 3
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: unwanted_products
    announcement_condition_value: 1
    assembly:
      shipment_count: 1
      stations: [as1]
      products:
        part_0:
          type: assembly_battery_blue
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 0]
        part_1:
          type: assembly_pump_blue
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]

models_over_bins:
   bin2:
     models:
       assembly_pump_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
   bin6:
     models:
       assembly_battery_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
faulty_products:
  - assembly_pump_green_1
  - assembly_battery_green_2

# don't forget to include at least 1 empty line here`

liuhaoran1, 2022-04-29:
`
options:
  insert_models_over_bins: true
  belt_population_cycles: 0
  insert_models_over_stations: false
  gazebo_state_logging: false
  current_gripper_type: gripper_tray
time_limit: 500

table_tray_infos:
  table_1: 
    tray_model: movable_tray_metal_shiny
    quantity: 1
  table_2: 
    tray_model: movable_tray_dark_wood
    quantity: 1

# mandatory: location of agvs and products at start
agv_infos:
  agv1:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0, 0]
          rpy: [0, 0, 'pi/4']
  agv2:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0.2, 0]
          rpy: [0, 0, 'pi/4']
  agv3:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0, 0]
          rpy: [0, 0, 0]
  agv4:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0.1, 0]
          rpy: [0, 0, 0] 

# Only one order 
orders:
  order_0:
    priority: 1
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: time
    announcement_condition_value: 0.0
    assembly:
      shipment_count: 1
      stations: [as4]
      products:
        part_0:
          type: assembly_battery_red
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 'pi']
        part_1:
          type: assembly_pump_red
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]
  order_1:
    priority: 3
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: unwanted_products
    announcement_condition_value: 1
    assembly:
      shipment_count: 1
      stations: [as1]
      products:
        part_0:
          type: assembly_battery_blue
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 0]
        part_1:
          type: assembly_pump_blue
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]

models_over_bins:
   bin2:
     models:
       assembly_pump_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
   bin6:
     models:
       assembly_battery_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
faulty_products:
  - assembly_pump_green_1
  - assembly_battery_green_2

liuhaoran1, 2022-04-29:
`
options:
  insert_models_over_bins: true
  belt_population_cycles: 0
  insert_models_over_stations: false
  gazebo_state_logging: false
  current_gripper_type: gripper_tray
time_limit: 500

table_tray_infos:
  table_1: 
    tray_model: movable_tray_metal_shiny
    quantity: 1
  table_2: 
    tray_model: movable_tray_dark_wood
    quantity: 1

agv_infos:
  agv1:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0, 0]
          rpy: [0, 0, 'pi/4']
  agv2:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0.2, 0]
          rpy: [0, 0, 'pi/4']
  agv3:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0, 0]
          rpy: [0, 0, 0]
  agv4:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0.1, 0]
          rpy: [0, 0, 0] 

orders:
  order_0:
    priority: 1
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: time
    announcement_condition_value: 0.0
    assembly:
      shipment_count: 1
      stations: [as4]
      products:
        part_0:
          type: assembly_battery_red
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 'pi']
        part_1:
          type: assembly_pump_red
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]
  order_1:
    priority: 3
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: unwanted_products
    announcement_condition_value: 1
    assembly:
      shipment_count: 1
      stations: [as1]
      products:
        part_0:
          type: assembly_battery_blue
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 0]
        part_1:
          type: assembly_pump_blue
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]

models_over_bins:
   bin2:
     models:
       assembly_pump_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
   bin6:
     models:
       assembly_battery_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
faulty_products:
  - assembly_pump_green_1
  - assembly_battery_green_2

liuhaoran1, 2022-04-29:
```

options:
  insert_models_over_bins: true
  belt_population_cycles: 0
  insert_models_over_stations: false
  gazebo_state_logging: false
  current_gripper_type: gripper_tray
time_limit: 500

table_tray_infos:
  table_1: 
    tray_model: movable_tray_metal_shiny
    quantity: 1
  table_2: 
    tray_model: movable_tray_dark_wood
    quantity: 1

agv_infos:
  agv1:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0, 0]
          rpy: [0, 0, 'pi/4']
  agv2:
    location: as1
    products:
      part_0:
        type: assembly_battery_blue
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 'pi/4']
      part_1:
        type: assembly_pump_blue
        pose:
          xyz: [0.0, 0.2, 0]
          rpy: [0, 0, 'pi/4']
  agv3:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [-0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0, 0]
          rpy: [0, 0, 0]
  agv4:
    location: as4
    products:
      part_0:
        type: assembly_pump_red
        pose:
          xyz: [0.15, -0.1, 0]
          rpy: [0, 0, 0]
      part_1:
        type: assembly_battery_red
        pose:
          xyz: [0.15, 0.1, 0]
          rpy: [0, 0, 0] 

orders:
  order_0:
    priority: 1
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: time
    announcement_condition_value: 0.0
    assembly:
      shipment_count: 1
      stations: [as4]
      products:
        part_0:
          type: assembly_battery_red
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 'pi']
        part_1:
          type: assembly_pump_red
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]
  order_1:
    priority: 3
    kitting_robot_health: 1 
    assembly_robot_health: 1 
    announcement_condition: unwanted_products
    announcement_condition_value: 1
    assembly:
      shipment_count: 1
      stations: [as1]
      products:
        part_0:
          type: assembly_battery_blue
          pose:
            xyz: [-0.032465, 0.174845, 0.15]
            rpy: [0, 0, 0]
        part_1:
          type: assembly_pump_blue
          pose:
            xyz: [0.032085, -0.152835, 0.25]
            rpy: [0, 0, 0]

models_over_bins:
   bin2:
     models:
       assembly_pump_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
   bin6:
     models:
       assembly_battery_green:
        xyz_start: [0.2, 0.2, 0.0]
        xyz_end: [0.4, 0.4, 0.0]
        rpy: [0, 0, 'pi/3']
        num_models_x: 2
        num_models_y: 2
faulty_products:
  - assembly_pump_green_1
  - assembly_battery_green_2

```

liuhaoran1, 2022-04-29:
> For 4. This has been fixed.
I didn't get the corresponding update.

closed 2022-05-27 by zeidk