usnistgov/ARIAC #156: locked movable tray falls off agv with new update

db7296, 2022-04-29:
With the new code, moving an agv causes the tray to fall off the agv.  The call to lock the tray returned with an ambiguous result, as there had been no previous call to lock the tray and the returned message is that the tray was already locked.

[INFO] [1651252304.942284, 37.714000]: waiting for lock_movable_tray service /ariac/kit_tray_1/lock
[INFO] [1651252304.944533, 37.715000]: calling lock_movable_tray service /ariac/kit_tray_1/lock
lock result:  success: False
message: "Movable tray is already locked on the kit tray"

https://user-images.githubusercontent.com/83300520/165991674-075c0b51-4f95-4236-8475-ad1cc722e0b1.mp4

zeidk, 2022-04-29:
Thanks, this should be a quick fix. Give me 1h or so.

db7296, 2022-04-30:
move_to_station service call

db7296, 2022-05-31:
This is no longer a move I try to make.

closed 2022-05-31 by db7296