usnistgov/ARIAC #158: When disable development mode, gripper change is impossible

mqt0029, 2022-04-30:
When launching using `roslaunch`, without the `--development-mode` flag, gripper change will always returns a `success: False`. The attached image is shown the gantry at the exact same location, one with development mode off, and the later on. Can't determine on test_competitor code since it depends on the `--development-mode` flag.

[image]

zeidk, 2022-05-01:
Thanks, I will take a look now.

zeidk, 2022-05-01:
Should be fixed in latest update.

closed 2022-05-01 by zeidk

mqt0029, 2022-05-01:
Confirmed. Working without `--development-mode` flag. Thank you!