usnistgov/ARIAC #160: Does the competitor's code have to be publicly available?

SyllogismRXS, 2022-05-02:
Based on the instructions for automatic evaluation via docker, it seems that each competitor's code needs to be publicly available since `git clone` is used to pull the code before building it inside of a docker image. Is publicly available code a requirement for this competition? 

Likewise, NIST has provided competitor's with Google Drive links to upload their code, but we don't know how this code will be used and/or configured during the automatic evaluation, which makes it difficult to replicate the automatic evaluation locally.

jaybrecht, 2022-05-03:
The code does not need to be publicly available it just needs to be accessible using the command line. This can be done through any means that you want. 

If you want to use a github private repository you can allow access using a personal access token. The instructions for setting that up can be seen here [(creating a personal access token)](https://docs.github.com/en/authentication/keeping-your-account-and-data-secure/creating-a-personal-access-token). 

You would then change the clone line in build_team_system.bash to `git clone https://<your_username>:<token>@github.com/<your_username/your_repository.git>`.

The only files that should be added to google drive folder are the three specified in the [automated evaluation wiki page](https://github.com/usnistgov/ARIAC/blob/ariac2022/wiki/documentation/automated_evaluation.md), which are
- team_config.yaml
- build_team_system.bash
- run_team_system.bash

SyllogismRXS, 2022-05-06:
Thanks! That's helpful and I was able to get it working locally in docker. It would be good if the mention of the personal access tokens was added to the automated evaluation instructions where these files are described.

zeidk, 2022-05-27:
Thank you, I will add it to the documentation.

closed 2022-05-27 by zeidk