usnistgov/ARIAC #161: rosservice call /ariac/briefcase_{N}/get_assembled_parts report

db7296, 2022-05-02:
rosservice call /ariac/briefcase_{N}/get_assembled_parts
returns regulators within a briefcase whether or not they are locked into place, as long as they are close to the port.  It does not return the presence of a sensor, even if it is locked in place.

It reports pumps and batteries correctly.

mqt0029, 2022-05-06:
Could be relating to #148. It seems like the submission procedure relies on the same topic for part count? Not entirely sure.

zeidk, 2022-05-10:
I assume the sensor is not reported in the scoring as well. Is the issue always with the regulator? Did you notice if it with a specific color for the regulator? I will check the collision with the regular models and I will report back.

mqt0029, 2022-05-10:
That is correct, as in there are scenarios where all parts are assembled. And if submission works, sometimes parts will not register as well (part is present but not counting towards correct type and correct color), sometimes it will register but not for bonus (pose is verified to be correct but not count toward pose points CP and bonus CC & CT & CP), and sometimes it will register correctly.

For context, I used red regulator and green sensor for the majority of my tests.

db7296, 2022-05-10:
The original question was whether there is a way to determine if a part is locked.  This only applies to the sensor and regulator.  Without a way to know if the part, especially the sensor, is locked, it is hard to determine when to release the part from the gripper.  So this is also an important consideration in the parameters returned.

db7296, 2022-05-13:
It's important to know if this service is going to work for determining if a regulator or sensor is locked into place.  There is no other way to determine that and, once locked, the part can no longer be moved.

As we near the end of the competition, it would be very helpful to know if this service will actually be available or not for reporting the locked state.

closed 2022-05-27 by zeidk