usnistgov/ARIAC #163: backwards battery?

db7296, 2022-05-03:
The sample trial config file "dev/test_hpo.yaml" has a product order that installs the battery backwards in the briefcase.  Seems like a smart,, agile robot would recognize this as a mistake in the order and install the battery properly rather than in a bad configuration that would possibly damage the respirator.  Might we encounter that case in the evaluation run?  If so, should the robot do the right thing or just blindly follow the faulty order?

zeidk, 2022-05-06:
Good question but we could imagine a new ventilator model where the battery's yaw is pi instead of 0. The gantry has to be able to handle this new orientation. You should also not assume the pump yaw will always be the same. The only parts which will always have the same position and the same orientation are the regulator and the sensor.

db7296, 2022-05-09:
Sounds good.  Thanks for the response.

closed 2022-05-09 by db7296