usnistgov/ARIAC #167: priority orders

db7296, 2022-05-08:
The wiki says, "High-priority orders will show up on the topic /ariac/orders as either order_1 or as order_0_update."  Does this mean that **all** orders with an order.id = order_1 or order.id= order_0_update are priority 3?  If not, is there any parameter passed in the order message to indicate priority?  I don't see that in the message structure, but just want to be sure.  Also, is it possible to get an order_1_update?

zeidk, 2022-05-10:
Hi,
the way we specify a high-priority order is in the configuration file (YAML) where the priority is set to 3. On /ariac/orders this priority does not show up but the order ID should tell you about the priority (order_1 and order_0_update). The rule of thumb is to address order_1 or order_0_update as soon it is announced.

No, you will never see order_1_update.

db7296, 2022-05-10:
OK, thanks for clarifying that all oriders with order ID = order_1 or order_0_update are priority 3.