usnistgov/ARIAC #170: latest build, path planning mostly just fails

db7296, 2022-05-25:
With the latest build, the gantry arm seems unable to support its own weight.  Many planning paths fail with:
[ERROR] [1653443788.919359846, 98.485000000]: RRTConnect: Unable to sample any valid states for goal tree

db7296, 2022-05-25:
Further testing-- rqt can move the arm OK, but moveit execution for the gantry arm fails in many, but not all, cases.

db7296, 2022-05-25:
Looks like the controllers stop and start a lot.  The moves are getting preempted.  Here is some console output:

You can start planning now!

[ INFO] [1653503153.833726937, 36.194000000]: 
------------------------
    Announcing Order
------------------------
-Order ID: order_0
-Priority: REGULAR
-Configuration File: qual_b.yaml
------------------------
-Kitting Shipment: x0

-Assembly Shipment: x1
  -Shipment #1
    -Assembly Station: as2
    -Products:
      assembly_regulator_red[(-0.222,-0.164,0.212),(-2.47956e-36,1.5708,0)]
      assembly_sensor_green[(0.405,0.164,0.11),(1.5708,-0,0)]
      assembly_battery_green[(-0.03,0.18,0.088),(0,-0,0)]
      assembly_pump_blue[(0.032,-0.153,0.076),(0,-0,0)]
------------------------

[ INFO] [1653503157.590627071, 37.005000000]: Stopped kitting controllers
[ INFO] [1653503157.610110988, 37.009000000]: Stopped gantry controllers
[ INFO] [1653503157.631572625, 37.014000000]: Started kitting controllers
[ INFO] [1653503157.654033108, 37.017000000]: Started gantry controllers
[ INFO] [1653503162.123730233, 38.004000000]: Stopped kitting controllers
[ INFO] [1653503162.142775495, 38.007000000]: Stopped gantry controllers
[ INFO] [1653503162.166940773, 38.012000000]: Started kitting controllers
[ INFO] [1653503162.193420291, 38.016000000]: Started gantry controllers
[ INFO] [1653503166.717271793, 39.003000000]: Stopped kitting controllers
[ INFO] [1653503166.740134125, 39.007000000]: Stopped gantry controllers
[ INFO] [1653503166.762299762, 39.012000000]: Started kitting controllers
[ INFO] [1653503166.788279233, 39.018000000]: Started gantry controllers
[ INFO] [1653503170.785718561, 39.885000000]: Received event 'stop'
[ INFO] [1653503171.063696053, 39.942000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1653503171.063892878, 39.942000000]: Preempt requested before the goal is planned and executed.
[ INFO] [1653503171.064174465, 39.942000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1653503171.064268342, 39.942000000]: Planning attempt 1 of at most 1
[ INFO] [1653503171.065918205, 39.942000000]: Planner configuration 'gantry_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1653503171.066602403, 39.942000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1653503171.076550741, 39.944000000]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1653503171.076704628, 39.945000000]: Solution found in 0.010341 seconds
[ INFO] [1653503171.100008929, 39.949000000]: SimpleSetup: Path simplification took 0.023156 seconds and changed from 3 to 2 states
[ INFO] [1653503171.104373682, 39.950000000]: Disabling trajectory recording
[ INFO] [1653503171.159661855, 39.962000000]: Stopping execution due to preempt request
[ INFO] [1653503171.159945523, 39.962000000]: Cancelling execution for /ariac/gantry/gantry_arm_controller
[ INFO] [1653503171.160045033, 39.962000000]: Stopped trajectory execution.
[ INFO] [1653503171.160542831, 39.962000000]: Controller /ariac/gantry/gantry_arm_controller successfully finished
[ INFO] [1653503171.160620642, 39.962000000]: Completed trajectory execution with status PREEMPTED ...
[ INFO] [1653503171.161020321, 39.962000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1653503171.161112703, 39.962000000]: Planning attempt 1 of at most 1
[ INFO] [1653503171.161857860, 39.962000000]: Planner configuration 'gantry_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1653503171.162124407, 39.962000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1653503171.185293035, 39.967000000]: RRTConnect: Created 7 states (4 start + 3 goal)
[ INFO] [1653503171.185341777, 39.967000000]: Solution found in 0.023359 seconds
[ INFO] [1653503171.237840150, 39.977000000]: SimpleSetup: Path simplification took 0.052441 seconds and changed from 4 to 2 states
[ INFO] [1653503171.364653701, 40.003000000]: Stopped kitting controllers
[ INFO] [1653503171.384718090, 40.006000000]: Controller /ariac/gantry/gantry_arm_controller successfully finished
[ WARN] [1653503171.384829161, 40.006000000]: Controller handle /ariac/gantry/gantry_arm_controller reports status PREEMPTED
[ INFO] [1653503171.384902404, 40.006000000]: Completed trajectory execution with status PREEMPTED ...
[ INFO] [1653503171.385062084, 40.006000000]: Stopped gantry controllers
[ INFO] [1653503171.404190962, 40.007000000]: Started kitting controllers
[ INFO] [1653503171.422618181, 40.012000000]: Started gantry controllers
[ INFO] [1653503175.957629291, 41.004000000]: Stopped kitting controllers
[ INFO] [1653503175.979434796, 41.005000000]: Stopped gantry controllers
[ INFO] [1653503175.998214134, 41.007000000]: Started kitting controllers
[ INFO] [1653503176.018848319, 41.011000000]: Started gantry controllers
[... 297 lines cut ...]
[ INFO] [1653503347.538152377, 77.005000000]: Stopped kitting controllers
[ INFO] [1653503347.562274953, 77.009000000]: Stopped gantry controllers
[ INFO] [1653503347.562583084, 77.009000000]: Controller /ariac/gantry/gantry_arm_controller successfully finished
[... cut ...]
[ INFO] [1653503357.096100552, 79.008000000]: Stopped gantry controllers
[ INFO] [1653503357.116168033, 79.012000000]: Started kitting controllers
[ INFO] [1653503357.136101617, 79.016000000]: Started gantry controllers
[ INFO] [1653503361.931908953, 80.005000000]: Stopped kitting controllers
[ INFO] [1653503361.954116467, 80.008000000]: Stopped gantry controllers
[ INFO] [1653503361.975395744, 80.012000000]: Started kitting controllers
[ INFO] [1653503361.997272776, 80.015000000]: Started gantry controllers

db7296, 2022-05-25:
> I received the attached videos with your email. This seems to be a moveit
> issue and not a gear issue. I will check if anything was modified in
> gantry_moveit_config and I will see if I encounter this problem.
>
> I tried running assembly_commander_node.py on qual_b.yaml? Besides the
> sensor bumping into the briefcase (which seems to be an offset issue on our
> end) I did not encounter the problem you reported. Could you try running
> assembly_commander_node.py on qual_b.yaml and see what happens?

OK, I'll try that.  I'm wondering if the moves are being
pre-empted somehow.  The server console output has tons of repeated stops
and starts of the controllers:

[ INFO] [1653503232.940243175, 53.062000000]: Planner configuration
'gantry_full' will use planner 'geometric::RRTConnect'. Additional
configuration parameters will be set when the planner is constructed.
[ INFO] [1653503232.940586531, 53.063000000]: RRTConnect: Starting planning
with 1 states already in datastructure
[ INFO] [1653503232.955718299, 53.066000000]: RRTConnect: Created 4 states
(2 start + 2 goal)
[ INFO] [1653503232.955775262, 53.066000000]: Solution found in 0.015362
seconds
[ INFO] [1653503232.975415144, 53.070000000]: SimpleSetup: Path
simplification took 0.019575 seconds and changed from 3 to 2 states
[ INFO] [1653503237.302040669, 54.002000000]: Stopped kitting controllers
[ INFO] [1653503237.321385890, 54.005000000]: Stopped gantry controllers
[ INFO] [1653503237.321579514, 54.006000000]: Controller
/ariac/gantry/gantry_arm_controller successfully finished
[ WARN] [1653503237.322155843, 54.006000000]: Controller handle
/ariac/gantry/gantry_arm_controller reports status PREEMPTED
[ INFO] [1653503237.345001998, 54.010000000]: Started kitting controllers
[ INFO] [1653503237.365893325, 54.014000000]: Started gantry controllers
[ INFO] [1653503242.014498373, 55.005000000]: Stopped kitting controllers
[ INFO] [1653503242.038270025, 55.009000000]: Stopped gantry controllers
[ INFO] [1653503242.059943044, 55.011000000]: Started kitting controllers
[ INFO] [1653503242.079564032, 55.013000000]: Started gantry controllers
[ INFO] [1653503246.739461407, 56.003000000]: Stopped kitting controllers
[ INFO] [1653503246.762380029, 56.006000000]: Stopped gantry controllers
[ INFO] [1653503246.785632917, 56.009000000]: Started kitting controllers
[ INFO] [1653503246.809541328, 56.012000000]: Started gantry controllers
[ INFO] [1653503251.429426784, 57.004000000]: Stopped kitting controllers

db7296, 2022-05-25:
I figured it out.

In my code, I send joint commands to moveit with the "wait" flag set to
False:

result = group.go(joint_goal, wait=False)

and then, if I want to wait for the joints to get to the goal, I use a
customized wait routine, which watches the various joint angles and
overall position.

If the execution is failing, I have a number of ways embedded in the
wait routine that help get the plan to succeed.

It seems that the new code is pre-empting the move or perhaps sending
some other null move that causes my move to be canceled.

I guess this works in general since most of the time people code the
moves with the wait flag set to True.

However, it kills my method.  This is a big problem because my custom
wait routine is really really good at rescuing failing plans.

db7296, 2022-05-25:
weird, I was about to post the above in the issues page and it just popped
up :)

zeidk, 2022-05-25:
OOOOooooh, the start and stop controllers output you see are part of the robot breakdown challenge. That should not happen unless we have the robot breakdown challenge in the yaml file. Did you change qual_b.yaml or are you using the one that was uploaded on Github? I ran a few tests earlier today using qual_b.yaml and I did not get these outputs. I will double check on my side.

zeidk, 2022-05-25:
One more thing, while your code is running, do `rostopic echo /ariac/robot_health` and let me know what outputs you get.

db7296, 2022-05-25:
rostopic echo /ariac/robot_health
kitting_robot_health: True
assembly_robot_health: True

The qual_b.yaml file is the same as the qualifiers version.

Still getting lots of the controller messages, but the code is managing to run the arm with the moveit flag set to True.  However, this is not the performance I programmed, as it is now bypassing my custom wait routines.

[ INFO] [1653511576.866723819, 19.003000000]: Stopped kitting controllers
[ INFO] [1653511576.888605401, 19.007000000]: Stopped gantry controllers
[ INFO] [1653511576.909438278, 19.010000000]: Started kitting controllers
[ INFO] [1653511576.929335179, 19.013000000]: Started gantry controllers
[ INFO] [1653511581.583235610, 20.005000000]: Stopped kitting controllers
[ INFO] [1653511581.606590492, 20.008000000]: Stopped gantry controllers
[ INFO] [1653511581.629621090, 20.011000000]: Started kitting controllers
[ INFO] [1653511581.652252716, 20.015000000]: Started gantry controllers
[ INFO] [1653511586.332885108, 21.003000000]: Stopped kitting controllers
[ INFO] [1653511586.358348678, 21.007000000]: Stopped gantry controllers
[ INFO] [1653511586.383898036, 21.009000000]: Started kitting controllers
[ INFO] [1653511586.408667595, 21.011000000]: Started gantry controllers
[ INFO] [1653511591.112226802, 22.003000000]: Stopped kitting controllers

db7296, 2022-05-25:
Dang, it is still failing sometimes, even with the wait flag set to True.

Robot health is still True

Still lots of controllers messages

zeidk, 2022-05-25:
Here is the probem, you shouldn't see true or false anymore on robot_health, you should see 'active' or 'inactive'. Do me a favor, do a catkin clean -y && catkin build. You need to regenerate the .h files for the RobotHealth.msg.

db7296, 2022-05-25:
That build failed, so I went back and tried to fully re-install ARIAC, but the build now fails:

dbarry@msi:~/ariac_ws$ catkin build
-----------------------------------------------------------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/dbarry/cmake-3.12.2-Linux-x86_64:/home/dbarry/ariac_2022/devel:/home/dbarry/littlebot/devel:/home/dbarry/neel_pi/devel:/home/dbarry/rpi/devel:/home/dbarry/ariac_ws/devel:/opt/ros/melodic:/home/dbarry/cmake-3.12.2-Linux-x86_64:/home/dbarry/cmake-3.12.2-Linux-x86_64:/home/dbarry/cmake-3.12.2-Linux-x86_64
Workspace:                   /home/dbarry/ariac_ws
-----------------------------------------------------------------------------------------------------------------------------------
Build Space:        [exists] /home/dbarry/ariac_ws/build
Devel Space:        [exists] /home/dbarry/ariac_ws/devel
Install Space:      [unused] /home/dbarry/ariac_ws/install
Log Space:         [missing] /home/dbarry/ariac_ws/logs
Source Space:       [exists] /home/dbarry/ariac_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-----------------------------------------------------------------------------------------------------------------------------------
[build] Found '12' packages in 0.0 seconds.                                                                                       
[build] Updating package table.                                                                                                   
Starting  >>> catkin_tools_prebuild                                                                                               
Finished  <<< catkin_tools_prebuild                [ 2.1 seconds ]                                                                
Starting  >>> controller_switcher                                                                                                 
Starting  >>> gantry_moveit_config                                                                                                
Starting  >>> gazebo_dev                                                                                                          
Starting  >>> gazebo_msgs                                                                                                         
Starting  >>> kitting_moveit_config                                                                                               
Finished  <<< gantry_moveit_config                 [ 2.3 seconds ]                                                                
__________________________________________________________________________________________________________________________________
Errors     << controller_switcher:cmake /home/dbarry/ariac_ws/logs/controller_switcher/build.cmake.000.log                        
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "nist_gear" with
  any of the following names:

    nist_gearConfig.cmake
    nist_gear-config.cmake

  Add the installation prefix of "nist_gear" to CMAKE_PREFIX_PATH or set
  "nist_gear_DIR" to a directory containing one of the above files.  If
  "nist_gear" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

cd /home/dbarry/ariac_ws/build/controller_switcher; catkin build --get-env controller_switcher | catkin env -si  /home/dbarry/cmake-3.12.2-Linux-x86_64/bin/cmake /home/dbarry/ariac_ws/src/ARIAC/controller_switcher --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/dbarry/ariac_ws/devel/.private/controller_switcher -DCMAKE_INSTALL_PREFIX=/home/dbarry/ariac_ws/install; cd -
..................................................................................................................................
Failed     << controller_switcher:cmake            [ Exited with code 1 ]                                                         
Failed    <<< controller_switcher                  [ 2.3 seconds ]                                                                
Abandoned <<< gazebo_plugins                       [ Unrelated job failed ]                                                       
Abandoned <<< gazebo_ros                           [ Unrelated job failed ]                                                       
Abandoned <<< gazebo_ros_control                   [ Unrelated job failed ]                                                       
[... cut ...]
[build]   Warnings:  None.                                                                                                        
[build]   Abandoned: 5 packages were abandoned.                                                                                   
[build]   Failed:    1 packages failed.                                                                                           
[build] Runtime: 8.8 seconds total.                                                                                               
[build] Note: Workspace packages have changed, please re-source setup files to use them.

zeidk, 2022-05-26:
Just did another push. Try building it again please.

db7296, 2022-05-26:
Now it builds and the arm behaves normally.

closed 2022-05-26 by db7296