usnistgov/ARIAC #172: wrong briefcase score

db7296, 2022-05-26:
Filled a briefcase correctly, verified with service calls, get_assembled_parts and get_content.  Also, visually looks correct.  Scoring still says one pose is wrong.

zeidk, 2022-05-26:
Ok, the briefcase recognized 4 parts attached but the pose of the sensor does not seem to register correctly. Can you do assembly with only the sensor and see if the pose is reported as not correct. If the issue persists I will either try to get this fixed or have assembly scenarios without the sensor.

db7296, 2022-05-26:
OK, will try that.  For clarification, does get_assembled_parts only report present for the regulator and the sensor when they are locked in?  Or will it report present if those parts are simply within the briefcase or perhaps within some range of the required placement?  The expectation was that it would allow determination of the locked state.

zeidk, 2022-05-26:
It reports the parts that are locked in. It will not report parts that are touching the briefcase and not locked in. They have to be in their respective slots to be reported.

db7296, 2022-05-26:
Running with a single sensor, placed correctly, was scored correctly, type, color, and pose.

rosservice call /ariac/briefcase_1/get_assembled_parts
connected_products: 
  briefcase_id: "briefcase_1"
  products: 
    - 
      type: "assembly_sensor_green"
      is_faulty: False

  order_0
------------------------
   ...total order score = completion score x priority: [7]
   ...completion score: [7]
   ...priority: [1]
   ...time taken: [85.443]
------------------------
    order_0_assembly_shipment_0
------------------------
      ...completion score: [7]
         +2 x (nb of products with CT & CP): [2]
         +1 x (nb of products with CC & CT & CP): [1]
         +bonus (0 or 4): [4]
      ------------------------
      ...complete? [true]
      ...shipped? [true]
      ...correct assembly station? [true]
      ...faulty products? [false]
      ...missing products? [false]
      ...unwanted products? [false]
      ------------------------
      ...number of products shipped: [1]
      ...products with correct type (CT): [1]
      ...products with correct color (CC): [1]
efcase_id: ''
  products: []

  order_0
------------------------
   ...total order score = completion score x priority: [7]
   ...completion score: [7]
   ...priority: [1]
   ...time taken: [85.443]
------------------------
    order_0_assembly_shipment_0
------------------------
      ...completion score: [7]
         +2 x (nb of products with CT & CP): [2]
         +1 x (nb of products with CC & CT & CP): [1]
         +bonus (0 or 4): [4]
      ------------------------
      ...complete? [true]
      ...shipped? [true]
      ...correct assembly station? [true]
      ...faulty products? [false]
      ...missing products? [false]
      ...unwanted products? [false]
      ------------------------
      ...number of products shipped: [1]
      ...products with correct type (CT): [1]
      ...products with correct color (CC): [1]
      ...products with correct pose (CP): [1]

db7296, 2022-05-26:
It is the regulator that is not getting a pose score.  Here is a run with just a single regulator where the score comes back with the pose not counted.

rosservice call /ariac/briefcase_1/get_assembled_parts
connected_products: 
  briefcase_id: "briefcase_1"
  products: 
    - 
      type: "assembly_regulator_red"
      is_faulty: False

  order_0
------------------------
   ...total order score = completion score x priority: [0]
   ...completion score: [0]
   ...priority: [1]
   ...time taken: [83.096]
------------------------
    order_0_assembly_shipment_0
------------------------
      ...completion score: [0]
         +2 x (nb of products with CT & CP): [0]
         +1 x (nb of products with CC & CT & CP): [0]
         +bonus (0 or 4): [0]
      ------------------------
      ...complete? [true]
      ...shipped? [true]
      ...correct assembly station? [true]
      ...faulty products? [false]
      ...missing products? [false]
      ...unwanted products? [false]
      ------------------------
      ...number of products shipped: [1]
      ...products with correct type (CT): [1]
      ...products with correct color (CC): [1]
      ...products with correct pose (CP): [0]

db7296, 2022-05-26:
Regarding the idea of not including a part in the briefcase for the competition.  It is very very late to change the scoring parameters and I hope that does not happen.  A lot of work was done to get the sensors and regulators reliably locking into place and it would be very sad for that to not get used.  The get_assembled_parts service seems to accurately determine that the sensor and regulator are there and locked.  Perhaps that could be looked at manually and, if the parts register there, then they get full points for pose?

zeidk, 2022-05-26:
Can you give it another try with the newest push?

db7296, 2022-05-26:
Full score with regulator by itself. :)
I'll try it with 4 parts now.

db7296, 2022-05-26:
Full score with 4 parts.  Nice fix.

closed 2022-05-26 by db7296